google-deepmind/dm_control #2: Segmentation Fault on checking mujoco licences

wookayin, 2018-01-03:
Hi all, thanks for open-sourcing such a great toolkit!
I would like to share and report a bug or issue in which segmentation fault occurs, which may also be happening for others.

```python
>>> from dm_control import suite
>>> env = suite.load(domain_name="cartpole", task_name="swingup")
[1]    28081 segmentation fault (core dumped)  python
```

I found that it happens during verifying the mujoco license, say when we call `_maybe_register_license()` (and `mjlib.mj_activate`).

stacktrace on gdb:

```
#0  0x0000000000017946 in ?? ()
#1  0x00007ffff0a84fff in mjl_check () from $HOME/.mujoco/mjpro150/bin/libmujoco150.so
#2  0x00007ffff0a4c013 in mjl_activate () from $HOME/.mujoco/mjpro150/bin/libmujoco150.so
#3  0x00007ffff0a30e4d in mj_activate () from $HOME/mujoco/mjpro150/bin/libmujoco150.so
...
```

So the segfault happens when `mj_activate` from `libmujoco150.so` is invoked. I think it might be a problem of mujoco, not of this package; but maybe something can be wrong with auto-wrapping via ctypes. Plus, the test (`python setup.py test`) also fails with the same segmentation fault.

I verified that `mjkey.txt` exists and it worked well with mujoco C++ samples and OpenAI's mujoco-py.

Capri2014, 2018-01-04:
same thing with my system as Ubuntu 14.04.5 LTS

borgwang, 2018-01-05:
Same issue

yotam, 2018-01-09:
We haven't been able to reproduce the issue so far. @borgwang are your system details the same?

borgwang, 2018-01-09:
@yotam Exactly the same (64bit machine running Ubuntu 14.04 and anaconda python 3.6). The segfault raised when `mj_activate` from `libmujoco150.so` is invoked.

HenryZhou7, 2018-01-13:
Ubuntu Release 14.04, anaconda python 2.7 has the same issue here.
```
>>> from dm_control import suite
>>> env = suite.load(domain_name='cartpole', task_name='swingup')
Segmentation fault (core dumped)
```

wookayin, 2018-01-13:
I highly suspect that it might be a **glibc** compatibility issue. Ubuntu 14.04 and RHEL systems have very old version (say 2.19 and 2.17, respectively), and it doesn't happen in 16.04 (with glibc 2.23).

alimuldal, 2018-01-26:
@wookayin @borgwang Are you able to run the MuJoCo example binaries?

wookayin, 2018-01-26:
Yes, I am.

```
❯❯❯ cd $HOME/.mujoco/
❯❯❯ export LD_LIBRARY_PATH=$HOME/.mujoco/mjpro150/bin/
❯❯❯ ls mjkey.txt
mjkey.txt
❯❯❯ mjpro150/bin/test mjpro150/model/humanoid.xml xs 10

Comparison of original and saved model
 Max difference : 2.61e-06
 Field name     : body_mass

Simulation ..........
 Simulation time      : 0.35 s
 Realtime factor      : 28.21 x
 Time per step        : 0.071 ms
 Contacts per step    : 10
 Constraints per step : 32
 Degrees of freedom   : 27
```

saran-t, 2018-01-27:
Just had time to look at this: I've been able to reproduce the issue with Anaconda and Ubuntu 14 LTS.

The issue seems to stem from the fact that Anaconda brings its own libstdc++ that is newer than the system's library.

Are you able to see if the crash occurs if you don't use it inside an Anaconda environment?

wookayin, 2018-01-27:
Yes, it happens both on non-anaconda environment (plain virtualenv, all of 2.7 and 3.x) and anaconda environments.

```
❯❯❯ workon tf; which python
$HOME/.virtualenvs/tf/bin/python
❯❯❯ python --version
Python 3.4.3

❯❯❯ python -c 'from dm_control import suite; env = suite.load(domain_name="cartpole", task_name="swingup")'
[1]    12613 segmentation fault (core dumped)  python -c
```

The python binary originates from the system python3 (say /usr/bin/python3 installed with apt, pre-compiled using GCC 4.8.4 and possibly the old version of glibc/libstdc++).

saran-t, 2018-01-27:
I believe you suspect right: this seems most likely an issue with glibc version conflict between Anaconda and the old Ubuntu system library.

As a workaround, I've managed to get the demo script to run by:
- `apt-get install libxml2 libxslt1.1`
- `LD_PRELOAD=libstdc++.so.6:libxslt.so.1:libexslt.so.0:libxml2.so.2 python my_script.py`

Obviously this is a very hacky workaround. It might result in another crash elsewhere. It might lead to a subtle problem that doesn't result in a crash. YMMV.

*PS: Don't add the `LD_PRELOAD` to your `.bashrc`. Preferably, don't `export` it in your shell either.

saran-t, 2018-03-19:
Closing this issue for now since I think the workaround above should work. If not, feel free to comment and reopen.

closed 2018-03-19 by saran-t

wookayin, 2018-04-14:
@saran-t That workaround works for Ubuntu 14.04. Would you be available to come up with one for RHEL systems? (with glibc 2.17, quite old and legacy) The counterpart libraries of libxml2 and libxslt are installed (latest).

```
[1]    2739 illegal hardware instruction (core dumped)  LD_PRELOAD=libstdc++.so.6:libxslt.so.1:libexslt.so.0:libxml2.so.2 ipython
```

Well, but fundamentally, why should it happen? Is it due to the mujoco binary is only compatible with recent glibc? However, openai gym's control environment just works. It would be great for the `dm_control` library to not suffer from this issue (without a work-around; workaround is a workaround IMHO), so can we have this issue open until it is resolved? Thanks.

HenryZhou7, 2018-04-27:
I wonder if the workaround will actually do for Python 3.6
I am getting the following error when executing `from dm_control import suite`
`OSError: /u/henryzhou/.mujoco/mjpro150/bin/libmujoco150.so: undefined symbol: __glewBlitFramebuffer
`