google-deepmind/dm_control #3: Got "undefined symbol" When Importing suite

Hwhitetooth, 2018-01-03:
Hi, there!

I followed the installation instructions and installed the package successfully. But when I did `from dm_control import suite`, I got an error like this:
```python
>>> from dm_control import suite
WARNING: Logging before flag parsing goes to stderr.
W0103 17:33:16.675210 140177339770624 __init__.py:35] mjbindings failed to import mjlib and other functions. libmujoco.so may not be accessible.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/zeyu/anaconda3/envs/mujoco/lib/python3.5/site-packages/dm_control/suite/__init__.py", line 28, in <module>
    from dm_control.suite import acrobot
  File "/home/zeyu/anaconda3/envs/mujoco/lib/python3.5/site-packages/dm_control/suite/acrobot.py", line 26, in <module>
    from dm_control import mujoco
  File "/home/zeyu/anaconda3/envs/mujoco/lib/python3.5/site-packages/dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/home/zeyu/anaconda3/envs/mujoco/lib/python3.5/site-packages/dm_control/mujoco/engine.py", line 43, in <module>
    from dm_control.mujoco import index
  File "/home/zeyu/anaconda3/envs/mujoco/lib/python3.5/site-packages/dm_control/mujoco/index.py", line 94, in <module>
    from dm_control.mujoco.wrapper import util
  File "/home/zeyu/anaconda3/envs/mujoco/lib/python3.5/site-packages/dm_control/mujoco/wrapper/__init__.py", line 25, in <module>
    from dm_control.mujoco.wrapper.core import callback_context
  File "/home/zeyu/anaconda3/envs/mujoco/lib/python3.5/site-packages/dm_control/mujoco/wrapper/core.py", line 34, in <module>
    from dm_control.mujoco.wrapper.mjbindings import functions
  File "/home/zeyu/anaconda3/envs/mujoco/lib/python3.5/site-packages/dm_control/mujoco/wrapper/mjbindings/functions.py", line 22, in <module>
    mjlib = util.get_mjlib()
  File "/home/zeyu/anaconda3/envs/mujoco/lib/python3.5/site-packages/dm_control/mujoco/wrapper/util.py", line 96, in get_mjlib
    return ctypes.cdll.LoadLibrary(_get_full_path(raw_path))
  File "/home/zeyu/anaconda3/envs/mujoco/lib/python3.5/ctypes/__init__.py", line 429, in LoadLibrary
    return self._dlltype(name)
  File "/home/zeyu/anaconda3/envs/mujoco/lib/python3.5/ctypes/__init__.py", line 351, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /home/zeyu/.mujoco/mjpro150/bin/libmujoco150.so: undefined symbol: __glewBlitFramebuffer
```
I tried to test Mujoco independently by runing `$HOME/.mujoco/mjpro150/bin/simulate` and it worked well.
My OS is Ubuntu 16.04 and I am using Python 3.5.4.
Can anyone help me with this?

Thanks a lot!

saran-t, 2018-01-03:
Hi, I just ran into this myself. If you have an Nvidia graphics card, a workaround for the time being is:

First, make sure that you have GLFW and GLEW installed: `sudo apt-get install libglfw3 libglew2.0

Then, preload GLEW and OpenGL into the Python process. This can be done by running: `LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so.2.0:/usr/lib/nvidia-381/libGL.so python` (replacing `nvidia-381` with the version installed on your own machine as appropriate, you can run `nvidia-smi` to determine the driver version currently in use).

Do **not** set the `LD_PRELOAD` environment variable globally in your shell, though!

Hwhitetooth, 2018-01-04:
@saran-t Thanks for your reply! And your solution works perfectly, though I am not using a Nvidia card.
Do you know why preloading is necessary?

foolyc, 2018-01-04:
@saran-t I follow your solution, and it not works for me.

efagerho, 2018-01-04:
This didn't quite work for me on Ubuntu 16.04 LTS, but the following works:

```
sudo apt-get install libglew-dev libglfw3-dev
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/nvidia-384/libGL.so python
```

blackvitriol, 2018-01-04:
@foolyc: try 
`sudo apt install mesa-utils
ldd $(which glxinfo) | grep libGL.so
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so.1.13:/usr/lib/nvidia-384/libGL.so.1
`

anarkia7115, 2018-01-04:
@efagerho I follow your solution on ubuntu 16.04. It works! Thank you!

yotam, 2018-01-04:
Hi all, we've pushed [an update](https://github.com/deepmind/dm_control/commit/3b736812407d821ec5d48bdea08ac53e00d78d2e) for this issue. Please let us know if this fixes it for you.

huapohen, 2018-01-05:
@saran-t @efagerho It works for me.  Thank you!

yotam, 2018-01-08:
@Hwhitetooth I'm going to mark this as resolved (see comment above) but if anyone still has this problem after trying the update, please comment and we'll reopen.

closed 2018-01-08 by yotam

tinyzqh, 2020-11-02:
I see the same problem, because the pycharm can't load the LD_LIBRARY_PATH of .bashrc. 
I resolved this problem after starting PyCharm from the terminal.