google-deepmind/dm_control #4: How to render?

Hwhitetooth, 2018-01-04:
There is a `render` module but I don't know how to use it. Could you please provide some examples of how to render?

wwxFromTju, 2018-01-04:
you  can use this:

from dm_control import suite
import matplotlib.pyplot as plt
import numpy as np

max_frame = 90

width = 480
height = 480
video = np.zeros((90, height, 2 * width, 3), dtype=np.uint8)

# Load one task:
env = suite.load(domain_name="cartpole", task_name="swingup")

# Iterate over a task set:
for domain_name, task_name in suite.BENCHMARKING:
  env = suite.load(domain_name, task_name)

# Step through an episode and print out reward, discount and observation.
action_spec = env.action_spec()
time_step = env.reset()
while not time_step.last():
  for i in range(max_frame):
    action = np.random.uniform(action_spec.minimum,
                             action_spec.maximum,
                             size=action_spec.shape)
    time_step = env.step(action)
    video[i] = np.hstack([env.physics.render(height, width, camera_id=0),
                          env.physics.render(height, width, camera_id=1)])
    #print(time_step.reward, time_step.discount, time_step.observation)
  for i in range(max_frame):
    img = plt.imshow(video[i])
    plt.pause(0.01)  # Need min display time > 0.0.
    plt.draw()

yotam, 2018-01-04:
Hi @Hwhitetooth the `render` module used internally within `dm_control`. We'll add a README so this is clearer, sorry for any confusion. Thanks @wwxFromTju for the snippet - see also page 10 of the [tech report](https://arxiv.org/abs/1801.00690).

closed 2018-01-04 by yotam

yotam, 2018-01-15:
Hi all, we're currently working on adding more rendering options to the open-source codebase, including headless + software rendering support.

iTakeshi, 2018-01-16:
Hi @yotam , could you provide an ETA for headless rendering?

Superchicken1, 2018-03-06:
Hi @yotam  
Is there any progress in this?

joseph280, 2018-06-11:
Hi @yotam any progress?

alimuldal, 2018-10-12:
Support for headless rendering was added in 1a31a7ab5c356f72c90719e86d82e25f3625500d (currently Linux only).

wookayin, 2020-04-08:
TL;DR for who might find this by Googling:

```
env.physics.render()                               # [height, width, 3]
env.physics.render(height, width, camera_id=1)     # [height, width, 3]
```

Signature:
```
def render(self, height=240, width=320, camera_id=-1, overlays=(), 
           depth=False, segmentation=False, scene_option=None):    
```

rfernand2, 2022-05-24:
Hi @wwxFromTju, here is update #2 which uses matplotlilb's animation.FuncAnimation.  It seems to run at a steady 29 FPS (RTX 3090, Intel i9-9900k), but has the downside of having to restructure the code a bit to put each RL step into a callback.

```
# dm_control_env2.py: test dm_control creation of an RL environment
import time
import numpy as np
import matplotlib.pyplot as plt
import matplotlib.animation as animation
from dm_control import suite

# Load one task:
env = suite.load(domain_name="cartpole", task_name="swingup")

action_spec = env.action_spec()
time_step = env.reset()

width = 480
height = 480

fig = plt.figure()
video = np.zeros((1, height, 2 * width, 3), dtype=np.uint8)
im = plt.imshow(video[0], animated=True)

max_frame = 90
frame_count = 0
started = time.time()

def one_step(args):
    action = np.random.uniform(action_spec.minimum, action_spec.maximum, size=action_spec.shape)
    time_step = env.step(action)
    
    frame = np.hstack([env.physics.render(height, width, camera_id=0),
            env.physics.render(height, width, camera_id=1)])

    im.set_array(frame) 

    # update stats
    global frame_count, started

    frame_count += 1
    if frame_count % max_frame == 0:
        elapsed = time.time() - started
        print("frame: {}, elapsed: {:.2f} secs, fps: {:.0f}".format(frame_count, elapsed, max_frame/elapsed))
        started = time.time()

    return im,

ani = animation.FuncAnimation(fig, one_step, interval=0, blit=True, repeat=False)
plt.show()

```

saran-t, 2022-05-24:
Just wanting to clarify, is this a regression specifically on `dm_control==1.0.3.post1` and `mujoco==2.2.0`, as in things were working fine in version 1.0.2?

rfernand2, 2022-05-24:
Good question:  this is the first time I have tried it.  I also forgot to mention a likely candidate for the difference: I am running on **Windows 10**.