google-deepmind/dm_control #6: import MuJoCo, ImportError: cannot import name 'constants'

wwxFromTju, 2018-01-04:
Dear all:
   when I import MuJoCo, some error happen:
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-2-7bf9adbdec98> in <module>()
----> 1 import dm_control.mujoco

/Users/codeMan/Documents/py-lab/dm_control/dm_control/mujoco/__init__.py in <module>()
     16 """Mujoco implementations of base classes."""
     17
---> 18 from dm_control.mujoco.engine import action_spec
     19
     20 from dm_control.mujoco.engine import Camera

/Users/codeMan/Documents/py-lab/dm_control/dm_control/mujoco/engine.py in <module>()
     41
     42 from dm_control import render
---> 43 from dm_control.mujoco import index
     44
     45 from dm_control.mujoco import wrapper

/Users/codeMan/Documents/py-lab/dm_control/dm_control/mujoco/index.py in <module>()
     92 # Internal dependencies.
     93
---> 94 from dm_control.mujoco.wrapper import util
     95 from dm_control.mujoco.wrapper.mjbindings import sizes
     96 import numpy as np

/Users/codeMan/Documents/py-lab/dm_control/dm_control/mujoco/wrapper/__init__.py in <module>()
     21
     22
---> 23 from dm_control.mujoco.wrapper import mjbindings
     24
     25 from dm_control.mujoco.wrapper.core import callback_context

/Users/codeMan/Documents/py-lab/dm_control/dm_control/mujoco/wrapper/mjbindings/__init__.py in <module>()
     22 from absl import logging
     23
---> 24 from dm_control.mujoco.wrapper.mjbindings import constants
     25 from dm_control.mujoco.wrapper.mjbindings import enums
     26 from dm_control.mujoco.wrapper.mjbindings import sizes

ImportError: cannot import name 'constants'

My python version is Python 3.5.3, and I use macOs sierra 10.12.6

efagerho, 2018-01-04:
Did you import making it editable? This gave me the same problem. In other words, 

```
pip import -e <path_to_dm_control>
```

instead of

```
pip import <path_to_dm_control>
```

I tend to always use the former when trying out new libraries, so I can edit them and see how they work. However, if you make the library editable, it doesn't seem to generate the needed mjbindings.

wwxFromTju, 2018-01-04:
you mean install?

wwxFromTju, 2018-01-04:
@efagerho you mean install?

efagerho, 2018-01-04:
@wwxFromTju: Yes. Fixed it.

wwxFromTju, 2018-01-04:
@efagerho It also have this err. can't solve it.

yotam, 2018-01-04:
Hi @wwxFromTju it sounds like the ctypes wrappers may not have been generated for some reason. If it ran successfully, you should see `constants.py` (and others [imported in __init__.py](https://github.com/deepmind/dm_control/blob/b1ac0c842859ccbf1b244cd11462608c5cfff04b/dm_control/mujoco/wrapper/mjbindings/__init__.py#L24)) in /Users/codeMan/Documents/py-lab/dm_control/dm_control/mujoco/wrapper/mjbindings/`. How did you install the package the first time?

wwxFromTju, 2018-01-04:
@yotam now, I create a clean conda env to install it, fix this problem.

yotam, 2018-01-04:
Thanks for the update. I'm going to close the issue, though if you run into it again do reopen. @efagerho I believe the `-e` install failed due to the way the output directory is determined [here](https://github.com/deepmind/dm_control/blob/master/setup.py#L83). We'll look into making this more flexible.

closed 2018-01-04 by yotam

andrewliao11, 2018-01-07:
hi @wwxFromTju 
Which python version do you use?

wwxFromTju, 2018-01-16:
@akfidjeland I try 3.6 and 3.5.

HenryZhou7, 2018-01-22:
I am working with Python 2.7 and it is not working for me. Still, I am using `pip install <path to dm_control>`
It would still give me 
```
>>> from dm_control import suite
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "dm_control/suite/__init__.py", line 28, in <module>
    from dm_control.suite import acrobot
  File "dm_control/suite/acrobot.py", line 26, in <module>
    from dm_control import mujoco
  File "dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "dm_control/mujoco/engine.py", line 43, in <module>
    from dm_control.mujoco import index
  File "dm_control/mujoco/index.py", line 94, in <module>
    from dm_control.mujoco.wrapper import util
  File "dm_control/mujoco/wrapper/__init__.py", line 23, in <module>
    from dm_control.mujoco.wrapper import mjbindings
  File "dm_control/mujoco/wrapper/mjbindings/__init__.py", line 24, in <module>
    from dm_control.mujoco.wrapper.mjbindings import constants
ImportError: cannot import name constants
>>>
```

HenryZhou7, 2018-01-22:
@wwxFromTju What is a clean conda environment has to do with the installation? I thought the installation of dm_control is pretty much isolated from conda environment?

wwxFromTju, 2018-01-24:
@HenryZhou7 create a new environment

wwxFromTju, 2018-09-25:
@HenryZhou7 In the initial conda environment, I have installed some python packages. So there may be some version number issues.

wwxFromTju, 2018-09-25:
@nilaytufek create a new conda env

nilaytufek, 2018-09-27:
> @nilaytufek create a new conda env

no it did now work. But, anyway, I soved my problem. I copied all under "C:\ProgramData\Anaconda3\Lib\site-packages\dm_control\mujoco" under <your_local_git_clone>\dm_control\mujoco .. That is woked for me.

powderluv, 2019-04-06:
This happens if you "pip install" but then try to import the module form the github source dir. change to some other path than the github checkout and it will work.

DanielTakeshi, 2019-05-10:
@powderluv  Thanks! Didn't realize that was another potential source of error.

FrankJIE09, 2021-12-29:
Because the name of lib "dm_control"  is same to the dir "dm_control" and "dm_control/dm_control", it cite the dir's constants,but there does not have it . I rename the dir's name to solve the problem.

Aakash-Kots1, 2024-10-08:
@FrankJIE09  changing the name of the directory worked. I changed it from dm_control to dm_control1 and then ran the code and it worked. Thanks!