google-deepmind/dm_control #7: Extending mjbindings with custom mj extensions?

efagerho, 2018-01-04:
Has anyone at DM tried extending mjbindings with their own stuff. I tried to add my own header file to the list of headers setup.py that it scans from ~/.mujoco/mjpro150/includes. The header contains:

```
#pragma once

// this is a C-API
#if defined(__cplusplus)
extern "C" {
#endif

#include "mujoco.h"

// Test extension.
MJAPI void my_mj_test(const mjModel* m);

#if defined(__cplusplus)
}
#endif

```
 Now I get the following error as expected:
```
>>> from dm_control.mujoco.wrapper.mjbindings import functions
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/efagerho/anaconda3/envs/mjtest/lib/python3.6/site-packages/dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/home/efagerho/anaconda3/envs/mjtest/lib/python3.6/site-packages/dm_control/mujoco/engine.py", line 43, in <module>
    from dm_control.mujoco import index
  File "/home/efagerho/anaconda3/envs/mjtest/lib/python3.6/site-packages/dm_control/mujoco/index.py", line 94, in <module>
    from dm_control.mujoco.wrapper import util
  File "/home/efagerho/anaconda3/envs/mjtest/lib/python3.6/site-packages/dm_control/mujoco/wrapper/__init__.py", line 23, in <module>
    from dm_control.mujoco.wrapper import mjbindings
  File "/home/efagerho/anaconda3/envs/mjtest/lib/python3.6/site-packages/dm_control/mujoco/wrapper/mjbindings/__init__.py", line 32, in <module>
    from dm_control.mujoco.wrapper.mjbindings import functions
  File "/home/efagerho/anaconda3/envs/mjtest/lib/python3.6/site-packages/dm_control/mujoco/wrapper/mjbindings/functions.py", line 33, in <module>
    """
  File "/home/efagerho/anaconda3/envs/mjtest/lib/python3.6/ctypes/__init__.py", line 361, in __getattr__
    func = self.__getitem__(name)
  File "/home/efagerho/anaconda3/envs/mjtest/lib/python3.6/ctypes/__init__.py", line 366, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /home/efagerho/.mujoco/mjpro150/bin/libmujoco150.so: undefined symbol: my_mj_test
```

I was wondering what would be the easiest way to make the wrapper load my custom shared object if I dump that in the same directory as libmujoco150.so?

saran-t, 2018-01-04:
For now, I'd suggest using `ctypes.CDLL` to load in your `.so` library and use `ctypes` to look up the symbols.

efagerho, 2018-01-05:
Do you have any idea how to do this easily, since most of the code uses mjlib directly, which is the ctypes object you get when loading libmujoco150.so? In the generated functions.py, you directly write into mjlib.my_mj_test, so that explicitly seems to try to find the symbol in libmujoco150.so despite having loaded the extension shared object first, i.e. ctypes will not operate transitively here with respect to symbol resolution.

I did get this working though with the following hack of get_mjlib():

```
class Combiner(object):
  def __init__(self, instances):
    # Pick a name that isn't used in the wrapped instances.
    self._95135instances = instances

  def __getattr__(self, attr):
    for i in self._95135instances:
      try:
        return getattr(i, attr)
      except AttributeError as e:
        pass        
    return getattr(self._95135instances[0], attr)

def get_mjlib():
  ctypes.CDLL(ctypes.util.find_library("GL"), ctypes.RTLD_GLOBAL)
  ctypes.CDLL(ctypes.util.find_library("GLEW"), ctypes.RTLD_GLOBAL)
  mujoco_lib = ctypes.cdll.LoadLibrary(_get_full_path(DEFAULT_MJLIB_PATH))
  extensions_lib = ctypes.cdll.LoadLibrary('/home/efagerho/.mujoco/mjpro150/bin/libmj_ext
ensions.so')
  return Combiner([mujoco_lib, extensions_lib])
```

saran-t, 2018-01-05:
If you don't mind accessing your extension functions separately from Mujoco's functions, then I think the only modification required is to load `libmujoco150.so` with `RTLD_GLOBAL`.

In your own code, you would have to import dm_control first, so as to cause `libmujoco150.so` to be loaded, before using `ctypes.CDLL` to load your own library as `extensions_lib`. Symbol resolution should now go through properly.

You would then call your own functions through `extensions_lib` while continuing to access Mujoco's functions through our wrapper.

efagerho, 2018-01-05:
Yes, that should work, the downside is that I would have to write the bindings myself (though this is only moving the code from the generated functions.py to my own library). I'll play around with the different options to see what leads to minimal amount of divergence from the dm_control code base.

yuvaltassa, 2018-01-09:
By the way ☝️, if what you're trying to do is install a MuJoCo callback, do the following:

1. Compile your callback as a C library:
```c
#include "mujoco.h"
void my_callback(const mjModel* m, mjData* d) 
{
  // your code here
}
```
2. Then, in Python:
```python
import ctypes
from dm_control.mujoco import wrapper
MYLIB = ctypes.cdll.LoadLibrary(‘path/to/my_lib.so’)
wrapper.core.set_callback('mjcb_passive', MYLIB.my_callback)
```
Done! 💫

closed 2018-03-19 by alimuldal