google-deepmind/dm_control #11: new domain with STL mesh

foolyc, 2018-01-08:
how to create a new domain with STL mesh file?
when i tried to add mesh file to the domain xml file, Segmentation fault (core dumped) will come out.

yotam, 2018-01-09:
Assuming you have a mesh imported in the MJCF file as

    <asset>
      <mesh name="meshname"   file="filename.stl"/>
    </asset>

where the `.stl` filename is specified relative to the MJCF/XML file. For a quick check, try to load the MJCF file in the `simulate` example provided with MuJoCo Pro.

Once that all works, you can use the named constructor [`mujoco.Physics.from_xml_path`](https://github.com/deepmind/dm_control/blob/3b9bf0aa6c910c668e5bb0dc2414f00738d90bb1/dm_control/mujoco/engine.py#L353), passing the path to the MJCF file.

Alternatively, you can pass the MJCF file contents to [`mujoco.Physics.from_xml_string`](https://github.com/deepmind/dm_control/blob/3b9bf0aa6c910c668e5bb0dc2414f00738d90bb1/dm_control/mujoco/engine.py#L330) and use the `assets` argument to supply the `.stl` data.

foolyc, 2018-01-10:
yes, i tried my xml in simulate provided with MuJoCo Pro, it works well.

foolyc, 2018-01-11:
yes, i tried my xml(both with STL or with out STL) in simulate provided with MuJoCo Pro, it works well.
But the  `mujoco.Physics.from_xml_path` and  `mujoco.Physics.from_xml_string` fail.

alimuldal, 2018-01-22:
Sorry, didn't mean to close this. Could you share the problem STL mesh as well as the XML file?

alimuldal, 2018-03-23:
Closing this for now since we haven't heard back from the OP and we lack the XML and STL file needed to reproduce the segfault. Please reopen and provide the missing info if this is still an issue.

closed 2018-03-23 by alimuldal