google-deepmind/dm_control #13: Batched environments?

efagerho, 2018-01-09:
Has there been any plans to support batched environments, where the environment runs parallel simulations or is the idea to just let the user manage manage a set of environments?

yuvaltassa, 2018-01-09:
We will support batched step() functionality in the near future: single
environment, multiple states.
Watch this space.
​

Yuval Tassa <http://yuvaltassa.com/> | Research Scientist | Google
DeepMind | tassa@google.com

fleeb24, 2019-03-02:
Any update on this? This would be great - my multiprocessing hacks are a mess!

Danfoa, 2021-11-10:
Any updates on this? I want to use dm_controll but for many learning controller tasks, the time bottleneck comes from not batching environments for policy inference. ISAAC GYM is providing a large set of desirable training features (batched environments is one of them), however, they lack the core features of dm_control.

yuvaltassa, 2021-11-12:
Hi folks, sorry for the delay. This is a core feature that will be implemented in MuJoCo's new Python bindings, which will replace dm_control's (based on pybind11 rather than ctypes). Please follow the discussion [here](https://github.com/deepmind/mujoco/issues/36).

closed 2021-11-12 by yuvaltassa