google-deepmind/dm_control #19: ctypes constants.py not generated

HenryZhou7, 2018-01-22:
Here is what I did when trying to install the package.
```
git clone <the github link>
# Change the DEFAULT_HEADERS_DIR in setup.py to my local include file (There isn't such option as headers-dir. I would be grateful if someone can point it to me how to use it)
# Set environment variables to MJKEY_PATH=<local path>/mjkey.txt
# Set environment variables to MJLIB_PATH=<local path>/bin
# I verified using echo command and it shows accordingly.
pip install .
```
Then when I try to run the tester code. I encounter the following error.
```
>>> from dm_control import suite
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "dm_control/suite/__init__.py", line 28, in <module>
    from dm_control.suite import acrobot
  File "dm_control/suite/acrobot.py", line 26, in <module>
    from dm_control import mujoco
  File "dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "dm_control/mujoco/engine.py", line 43, in <module>
    from dm_control.mujoco import index
  File "dm_control/mujoco/index.py", line 94, in <module>
    from dm_control.mujoco.wrapper import util
  File "dm_control/mujoco/wrapper/__init__.py", line 23, in <module>
    from dm_control.mujoco.wrapper import mjbindings
  File "dm_control/mujoco/wrapper/mjbindings/__init__.py", line 24, in <module>
    from dm_control.mujoco.wrapper.mjbindings import constants
ImportError: cannot import name constants
```
I saw somewhere that it means that the mjbindings is not generated. But I use `pip install .` instead of `pip install -e`, so that shouldn't be the issue. 
Can someone please help me here?

alimuldal, 2018-01-22:
Hi Henry,

Are you by any chance trying to `import dm_control` from within the same directory where you checked out the git repository?

By default, `constants.py` and the other auto-generated files in `mjbindings` are created in `/path/to/site-packages/dm_control/mujoco/wrapper/mjbindings/`, rather than "in place"* within `./dm_control/mujoco/wrapper/mjbindings/` (where `.` is the directory you checked out the git repo to). If you launch a Python interpreter in `.` and then try to `import dm_python`, the interpreter will attempt to import from `./dm_control` rather than `site-packages/dm_control`, and since the version in `.` doesn't contain `mujoco/wrapper/mjbindings/constants.py` etc. you'll get an `ImportError`.

After installation, if you `cd` to a different directory that doesn't contain a subdirectory named `dm_control` then Python will import from `site-packages/dm_control`, which ought to succeed.

---

Regarding your other question:

> Change the DEFAULT_HEADERS_DIR in setup.py to my local include file (There isn't such option as headers-dir. I would be grateful if someone can point it to me how to use it)

`--headers-dir=` is an optional argument to a custom setuptools command (`python setup.py build_mjbindings`) that parses the MuJoCo headers and creates the autogenerated Python files in `mjbindings/`. It was intended to be used like so:

```bash
python setup.py build_mjbindings --headers-dir=/path/to/mujoco/include/dir/
```

Unfortunately this isn't very useful for you at the moment, since `python setup.py install` doesn't accept a `--headers-dir=` argument, and calling `install` after `build_mjbindings` will cause `build_mjbindings` to be run again without the custom `--headers-dir=` argument, which won't work if the default path is invalid. I'm currently working on a fix.

---

\* It's also possible to generate the bindings in place using `python setup.py build_mjbindings --inplace=True`, which we do in order to run the test suite without needing to install anything first.

alimuldal, 2018-01-26:
It's now possible to pass the `--headers-dir=` argument to `python setup.py install` if you want to specify a custom location for the MuJoCo header files at installation time. The import error you're seeing is probably due to attempting to import `dm_control` from within the source directory.

closed 2018-01-26 by alimuldal