google-deepmind/dm_control #20: No OpenGL rendering backend

HenryZhou7, 2018-01-22:
I am having the following trouble even though the requirement for installing `dm_control` is met.
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/suite/__init__.py", line 110, in load
    visualize_reward)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/suite/__init__.py", line 140, in build_environment
    env = domain.SUITE[task_name](**task_kwargs)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/suite/cartpole.py", line 66, in swingup
    physics = Physics.from_xml_string(*get_model_and_assets())
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/mujoco/engine.py", line 344, in from_xml_string
    return cls.from_model(model)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/mujoco/engine.py", line 327, in from_model
    return cls(data)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/mujoco/engine.py", line 120, in __init__
    self._reload_from_data(data)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/mujoco/engine.py", line 306, in _reload_from_data
    render_context = render.Renderer(_MAX_WIDTH, _MAX_HEIGHT)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/render/__init__.py", line 51, in Renderer
    raise ImportError('No OpenGL rendering backend could be imported.')
ImportError: No OpenGL rendering backend could be imported.
```

alimuldal, 2018-01-22:
Hi Henry, are you able to run the following?
```python
import glfw
assert glfw.init()
```

HenryZhou7, 2018-01-22:
Hi, 
I couldn't run the second command.
```
>>> import glfw
>>> assert glfw.init()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/glfw.py", line 717, in init
    res = _glfw.glfwInit()
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/glfw.py", line 626, in errcheck
    _reraise(exc[1], exc[2])
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/glfw.py", line 39, in _reraise
    raise (exception, None, traceback)
glfw.GLFWError: (65544) X11: The DISPLAY environment variable is missing
>>>
```

alimuldal, 2018-01-23:
That error suggests that you're running on a machine without an X display. GLFW requires X11, so it can't be used on headless servers. We're working on adding support for other OpenGL backends that support headless rendering.

HenryZhou7, 2018-01-23:
Yes. I am using `ssh` connection to run `dm_control` on the server. With `ssh -X <server>`, it would still give me an error.
```
>>> env = suite.load(domain_name="cartpole", task_name="swingup")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/suite/__init__.py", line 110, in load
    visualize_reward)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/suite/__init__.py", line 140, in build_environment
    env = domain.SUITE[task_name](**task_kwargs)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/suite/cartpole.py", line 66, in swingup
    physics = Physics.from_xml_string(*get_model_and_assets())
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/mujoco/engine.py", line 344, in from_xml_string
    return cls.from_model(model)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/mujoco/engine.py", line 327, in from_model
    return cls(data)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/mujoco/engine.py", line 120, in __init__
    self._reload_from_data(data)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/mujoco/engine.py", line 306, in _reload_from_data
    render_context = render.Renderer(_MAX_WIDTH, _MAX_HEIGHT)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/render/base.py", line 44, in __init__
    self._create(max_width, max_height)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/render/glfw_renderer.py", line 42, in _create
    _maybe_init_glfw()
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/render/glfw_renderer.py", line 33, in _maybe_init_glfw
    if not glfw.init():
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/glfw.py", line 717, in init
    res = _glfw.glfwInit()
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/glfw.py", line 626, in errcheck
    _reraise(exc[1], exc[2])
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/glfw.py", line 39, in _reraise
    raise (exception, None, traceback)
glfw.GLFWError: (65544) X11: Failed to open display localhost:15.0
```

Is there a way to turn off the graphical rendering and just interact with the environment? Thanks!

jiameij, 2018-01-26:
i couldn't run the second command

`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jiamj/anaconda3/lib/python3.6/site-packages/glfw/__init__.py", line 723, in init
    res = _glfw.glfwInit()
  File "/home/jiamj/anaconda3/lib/python3.6/site-packages/glfw/__init__.py", line 632, in errcheck
    _reraise(exc[1], exc[2])
  File "/home/jiamj/anaconda3/lib/python3.6/site-packages/glfw/__init__.py", line 35, in _reraise
    raise exception.with_traceback(traceback)
  File "/home/jiamj/anaconda3/lib/python3.6/site-packages/glfw/__init__.py", line 611, in callback_wrapper
    return func(*args, **kwargs)
  File "/home/jiamj/anaconda3/lib/python3.6/site-packages/glfw/__init__.py", line 785, in _raise_glfw_errors_as_exceptions
    raise GLFWError(message)
glfw.GLFWError: (65544) b'X11: The keyboard extension is not available'
Fatal Python error: Couldn't create autoTLSkey mapping

Thread 0x00007f1efe2c6740 (most recent call first):
Aborted (core dumped)
`
how can i solve this problem?
thanks~

alimuldal, 2018-01-26:
@HenryZhou7 Does the server you're connecting to have a GPU? If not then GLFW definitely won't work, even with X11 forwarding. In f87e7873ca0f0580442a10b6b1fa3d774ab0e462 I added the option to disable rendering support entirely by launching your Python interpreter with the environment variable `DISABLE_MUJOCO_RENDERING=1`, which should allow you to still use the MuJoCo bindings on machines where you're not able to create an OpenGL context provided that you don't make any rendering calls. We're hoping to add support for headless software rendering soon.

@jiameij That error in `glfw.init()` looks like https://bugs.python.org/issue7544, which can occur under low memory conditions.

HenryZhou7, 2018-01-26:
Somehow it would still give me the error.

```
$ echo $DISABLE_MUJOCO_RENDERING
1
$ python
Python 2.7.14 |Anaconda custom (64-bit)| (default, Oct 16 2017, 17:29:19) 
[GCC 7.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from dm_control import suite
>>> env = suite.load(domain_name="cartpole", task_name="swingup")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/suite/__init__.py", line 110, in load
    visualize_reward)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/suite/__init__.py", line 140, in build_environment
    env = domain.SUITE[task_name](**task_kwargs)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/suite/cartpole.py", line 66, in swingup
    physics = Physics.from_xml_string(*get_model_and_assets())
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/mujoco/engine.py", line 344, in from_xml_string
    return cls.from_model(model)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/mujoco/engine.py", line 327, in from_model
    return cls(data)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/mujoco/engine.py", line 120, in __init__
    self._reload_from_data(data)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/mujoco/engine.py", line 306, in _reload_from_data
    render_context = render.Renderer(_MAX_WIDTH, _MAX_HEIGHT)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/render/base.py", line 44, in __init__
    self._create(max_width, max_height)
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/render/glfw_renderer.py", line 42, in _create
    _maybe_init_glfw()
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/dm_control/render/glfw_renderer.py", line 33, in _maybe_init_glfw
    if not glfw.init():
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/glfw.py", line 717, in init
    res = _glfw.glfwInit()
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/glfw.py", line 626, in errcheck
    _reraise(exc[1], exc[2])
  File "/ais/gobi5/movie4d/conda_install/anaconda2/lib/python2.7/site-packages/glfw.py", line 39, in _reraise
    raise (exception, None, traceback)
glfw.GLFWError: (65544) X11: Failed to open display localhost:13.0
>>>
```
I have set the environment in `.bashrc` file and it is still not working.

alimuldal, 2018-01-26:
Did you remove your previous version of `dm_control` and install the most recent version from our repository?

LinxiFan, 2018-01-27:
When using dm_control within a docker image, I also have the same problem `No OpenGL rendering backend`

HenryZhou7, 2018-01-27:
@alimuldal I reinstalled the package and the following codes work on my system.
```
import os
os.environ['DISABLE_MUJOCO_RENDERING'] = '1'
from dm_control import suite
env = suite.load(domain_name="cartpole", task_name="swingup")
```
Thank you so much for the help!

alimuldal, 2018-01-29:
@LinxiFan Presumably your Docker image is also headless, so again GLFW won't work there. As a workaround you can set `DISABLE_MUJOCO_RENDERING=1` when launching your interpreter.

alimuldal, 2018-10-10:
We added an OSMesa-based rendering backend in 1a31a7ab5c356f72c90719e86d82e25f3625500d, which can be used to render on headless machines (currently only supported on Linux).

closed 2018-10-10 by alimuldal

ruifengma, 2019-02-25:
I also got this issue,  I'm using macos mojave
Traceback (most recent call last):
  File "/Users/ruifengma/Code/deepcontrol/soccer.py", line 5, in <module>
    env = dm_soccer.load(team_size=2, time_limit=10.)
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/dm_control/locomotion/soccer/__init__.py", line 75, in load
    time_limit=time_limit)
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/dm_control/composer/environment.py", line 305, in __init__
    strip_singleton_obs_buffer_dim=strip_singleton_obs_buffer_dim)
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/dm_control/composer/environment.py", line 210, in __init__
    self._recompile_physics()
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/dm_control/composer/environment.py", line 220, in _recompile_physics
    physics = self._make_physics()
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/dm_control/composer/environment.py", line 233, in _make_physics
    return mjcf.Physics.from_mjcf_model(self._task.root_entity.mjcf_model)
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/dm_control/mjcf/physics.py", line 471, in from_mjcf_model
    return cls.from_xml_string(xml_string=xml_string, assets=assets)
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/dm_control/mujoco/engine.py", line 344, in from_xml_string
    return cls.from_model(model)
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/dm_control/mujoco/engine.py", line 327, in from_model
    return cls(data)
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/dm_control/mujoco/engine.py", line 115, in __init__
    self._reload_from_data(data)
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/dm_control/mjcf/physics.py", line 506, in _reload_from_data
    super(Physics, self)._reload_from_data(data)
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/dm_control/mujoco/engine.py", line 295, in _reload_from_data
    self._make_rendering_contexts()
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/dm_control/mujoco/engine.py", line 423, in _make_rendering_contexts
    max_width=max_width, max_height=max_height)
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/dm_control/_render/__init__.py", line 79, in Renderer
    raise RuntimeError('No OpenGL rendering backend is available.')
RuntimeError: No OpenGL rendering backend is available.

while trying to execute import glfw, I got this error

Traceback (most recent call last):
  File "/Users/ruifengma/Code/deepcontrol/try.py", line 1, in <module>
    import glfw
  File "/Users/ruifengma/.virtualenvs/dm/lib/python3.6/site-packages/glfw/__init__.py", line 34, in <module>
    raise ImportError("Failed to load GLFW3 shared library.")
ImportError: Failed to load GLFW3 shared library.

alimuldal, 2019-02-25:
@ruifengma One possible reason for that error would be if the GLFW shared library is not in your `DYLD_LIBRARY_PATH` (see [here](https://github.com/deepmind/dm_control#additional-instructions-for-homebrew-users-on-macos))

GoingMyWay, 2019-05-13:
@ruifengma  

On macos, maybe you should `brew install glfw3`.