google-deepmind/dm_control #21: Getting `glfw.GLFWError`

LinxiFan, 2018-01-26:
When loading any task, I'm consistently getting 
```
glfw.GLFWError: (65543) b'GLX: Failed to create context: BadValue (integer parameter out of range for operation)'
Exception ignored in: <bound method Renderer.__del__ of <dm_control.render.glfw_renderer.GLFWRenderer object at 0x7f77ae4e8e48>>
Traceback (most recent call last):
  File "/home/anaconda3/lib/python3.6/site-packages/dm_control/render/base.py", line 110, in __del__
  File "/home/anaconda3/lib/python3.6/site-packages/dm_control/render/glfw_renderer.py", line 64, in free_context
AttributeError: 'GLFWRenderer' object has no attribute '_context'
```

My OS is Ubuntu 16.04, python 3.6. I installed `libglfw` from source as suggested by other users. I also installed `libglew1.13` because `libglew2.0` doesn't exist in apt-get on Ubuntu 16.04

alimuldal, 2018-01-26:
Are you able to run the following?
```python
import glfw
glfw.init()
```
If not then there may be a problem with your GLFW installation.

If you don't require rendering then you can disable rendering support by launching your Python interpreter with the environment variable `DISABLE_MUJOCO_RENDERING=1`. This should allow you to load and run tasks from the suite provided that you don't make any rendering calls.

LinxiFan, 2018-01-26:
Thanks for your help. Yes, I'm able to run `glfw.init()` and it returns `1`. 
I do need rendering support. Is there any other reason that might have caused the error message?

alimuldal, 2018-10-10:
As of 1a31a7ab5c356f72c90719e86d82e25f3625500d we support OSMesa on Linux, which can be used for rendering if GLFW is not available

closed 2018-10-10 by alimuldal