google-deepmind/dm_control #23: Reacher observation contains constant element

danijar, 2018-02-21:
```python
import numpy as np
from dm_control import suite

env = suite.load('reacher', 'easy')
action_spec = env.action_spec()
time_step = env.reset()
while not time_step.last():
  print(time_step.observation['to_target'][2])  # Always 0; possibly z coordinate?
  action = np.random.uniform(action_spec.minimum, action_spec.maximum, action_spec.shape)
  time_step = env.step(action)
```

alimuldal, 2018-03-23:
Is the constant element causing problems for you? I'm reluctant to remove it from the observation array since this would be a breaking change (e.g. for agents that assume a particular size and/or order of the observation array).

danijar, 2018-03-24:
It's not causing any problems, I just thought it was unintended. It might be a bit confusing to users since it's the only position vector that consists of 3 elements.

rajcscw, 2018-09-09:
What are the actions in the reacher environment? Is there any place where we can find any information about them?

alimuldal, 2018-10-19:
Fixed in e9a4ae59ec5a29338802c9382eb7634350b9e2c4

closed 2018-10-19 by alimuldal