google-deepmind/dm_control #24: Segmentation fault when importing dm_control before TensorFlow

danijar, 2018-03-01:
Importing TensorFlow and then dm_control works:

```sh
$ python3 -Xfaulthandler -c 'import tensorflow; from dm_control import suite; print("Done")'
Done
```

Importing dm_control and then TensorFlow (the alphabetical order) results in a segfault:

```sh
$ python3 -Xfaulthandler -c 'from dm_control import suite; import tensorflow; print("Done")'                                                                              
Fatal Python error: Segmentation fault

Current thread 0x00007f0b6fa72700 (most recent call first):
  File "<frozen importlib._bootstrap>", line 222 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 914 in create_module
  File "<frozen importlib._bootstrap>", line 577 in module_from_spec
  File "<frozen importlib._bootstrap>", line 666 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 693 in _load
  File "/usr/lib/python3.5/imp.py", line 342 in load_dynamic
  File "/usr/lib/python3.5/imp.py", line 242 in load_module
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/pywrap_tensorflow_internal.py", line 24 in swig_import_helper
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/pywrap_tensorflow_internal.py", line 28 in <module>
  File "<frozen importlib._bootstrap>", line 222 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 673 in exec_module
  File "<frozen importlib._bootstrap>", line 673 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/pywrap_tensorflow.py", line 58 in <module>
  File "<frozen importlib._bootstrap>", line 222 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 673 in exec_module
  File "<frozen importlib._bootstrap>", line 673 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "<frozen importlib._bootstrap>", line 222 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1016 in _handle_fromlist
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/__init__.py", line 49 in <module>
  File "<frozen importlib._bootstrap>", line 222 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 673 in exec_module
  File "<frozen importlib._bootstrap>", line 673 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/__init__.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 222 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 673 in exec_module
  File "<frozen importlib._bootstrap>", line 673 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "<string>", line 1 in <module>
zsh: segmentation fault  python3 -Xfaulthandler -c
```

I've replicated this on my Debian 9 machine (Google workstation) with Python versions 2.7, 3.5, 3.6 and TensorFlow versions 1.4.1 and 1.5.0 with GPU support. I'm using the current master branch of the repository.

WilsonWangTHU, 2018-06-28:
Hi @danijar , I am able to reproduce your errors here.

Hoping there is a fix soon...

alimuldal, 2018-10-28:
I believe this has the same root cause as #2, i.e. a libstdc++ ABI conflict. A workaround is to use `LD_PRELOAD` the correct version of `libstdc++`:
```bash
~$ python -c 'from dm_control import suite; import tensorflow; print("Done")'
Segmentation fault

~$ LD_PRELOAD=libstdc++.so.6 python -c 'from dm_control import suite; import tensorflow; print("Done")'
Done
```
It's also worth noting that only GLFW seems to be affected - I can also avoid the segfault by forcing the bindings to use the OSMesa (software) rendering backend instead of GLFW:
```bash
~$ MUJOCO_GL=osmesa python -c 'from dm_control import suite; import tensorflow; print("Done")'
Done
```

alimuldal, 2018-11-12:
Closed in e374a4a75461d07a80a1316463c45f9ccbb45324

closed 2018-11-12 by alimuldal