google-deepmind/dm_control #25: ImportError: cannot import name 'constants'

inksci, 2018-03-07:
When I run this:
```
from dm_control import suite
```
The error occurs:
```
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-2-2addf36a480e> in <module>()
----> 1 from dm_control import suite

~/ai-ink/Jupyter-zhouxin12/github/dm_control/dm_control/suite/__init__.py in <module>()
     26 from dm_control.rl import control
     27 
---> 28 from dm_control.suite import acrobot
     29 from dm_control.suite import ball_in_cup
     30 from dm_control.suite import cartpole

~/ai-ink/Jupyter-zhouxin12/github/dm_control/dm_control/suite/acrobot.py in <module>()
     24 # Internal dependencies.
     25 
---> 26 from dm_control import mujoco
     27 from dm_control.rl import control
     28 from dm_control.suite import base

~/ai-ink/Jupyter-zhouxin12/github/dm_control/dm_control/mujoco/__init__.py in <module>()
     16 """Mujoco implementations of base classes."""
     17 
---> 18 from dm_control.mujoco.engine import action_spec
     19 
     20 from dm_control.mujoco.engine import Camera

~/ai-ink/Jupyter-zhouxin12/github/dm_control/dm_control/mujoco/engine.py in <module>()
     41 
     42 from dm_control import render
---> 43 from dm_control.mujoco import index
     44 
     45 from dm_control.mujoco import wrapper

~/ai-ink/Jupyter-zhouxin12/github/dm_control/dm_control/mujoco/index.py in <module>()
     92 # Internal dependencies.
     93 
---> 94 from dm_control.mujoco.wrapper import util
     95 from dm_control.mujoco.wrapper.mjbindings import sizes
     96 import numpy as np

~/ai-ink/Jupyter-zhouxin12/github/dm_control/dm_control/mujoco/wrapper/__init__.py in <module>()
     21 
     22 
---> 23 from dm_control.mujoco.wrapper import mjbindings
     24 
     25 from dm_control.mujoco.wrapper.core import callback_context

~/ai-ink/Jupyter-zhouxin12/github/dm_control/dm_control/mujoco/wrapper/mjbindings/__init__.py in <module>()
     22 from absl import logging
     23 
---> 24 from dm_control.mujoco.wrapper.mjbindings import constants
     25 from dm_control.mujoco.wrapper.mjbindings import enums
     26 from dm_control.mujoco.wrapper.mjbindings import sizes

ImportError: cannot import name 'constants'
```

wuelfj, 2018-03-13:
I have the same problem, any ideas someone?

bzdlity, 2018-03-19:
I have the same problem when first cloning the repository and then  running pip install dm_control.
If I install this by using pip install git+git://github.com/deepmind/dm_control.git, the error would become 
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType.
Any ideas?
By the way, I'm using win10 , I think it might due to my operating system?

etom2deepmind, 2018-03-19:
Duplicate of #6

etom2deepmind, 2018-03-19:
This happens when the interpreter fails to find the auto-generated code, often because the interpreter starts in the source tree or when installation is run with "install e".

closed 2018-03-19 by etom2deepmind