google-deepmind/dm_control #27: TypeError: expected str, bytes or os.PathLike object, not NoneType

bzdlity, 2018-03-19:
I tried to install dm_control by pip install git+git://github.com/deepmind/dm_control.git.
However, I can import dm_control successfully, but when I used 'from dm_control import suite', error came.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\bzdli\Anaconda3\envs\drl\lib\site-packages\dm_control\suite\__init__.py", line 28, in <module>
    from dm_control.suite import acrobot
  File "C:\Users\bzdli\Anaconda3\envs\drl\lib\site-packages\dm_control\suite\acrobot.py", line 26, in <module>
    from dm_control import mujoco
  File "C:\Users\bzdli\Anaconda3\envs\drl\lib\site-packages\dm_control\mujoco\__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "C:\Users\bzdli\Anaconda3\envs\drl\lib\site-packages\dm_control\mujoco\engine.py", line 43, in <module>
    from dm_control.mujoco import index
  File "C:\Users\bzdli\Anaconda3\envs\drl\lib\site-packages\dm_control\mujoco\index.py", line 94, in <module>
    from dm_control.mujoco.wrapper import util
  File "C:\Users\bzdli\Anaconda3\envs\drl\lib\site-packages\dm_control\mujoco\wrapper\__init__.py", line 23, in <module>
    from dm_control.mujoco.wrapper import mjbindings
  File "C:\Users\bzdli\Anaconda3\envs\drl\lib\site-packages\dm_control\mujoco\wrapper\mjbindings\__init__.py", line 28, in <module>
    from dm_control.mujoco.wrapper.mjbindings import wrappers
  File "C:\Users\bzdli\Anaconda3\envs\drl\lib\site-packages\dm_control\mujoco\wrapper\mjbindings\wrappers.py", line 14, in <module>
    from dm_control.mujoco.wrapper import util
  File "C:\Users\bzdli\Anaconda3\envs\drl\lib\site-packages\dm_control\mujoco\wrapper\util.py", line 57, in <module>
    "~/.mujoco/mjpro150/bin", _get_shared_library_filename())
  File "C:\Users\bzdli\Anaconda3\envs\drl\lib\site-packages\dm_control\mujoco\wrapper\util.py", line 47, in _get_shared_library_filename
    libc_filename = os.path.split(libc_path)[1]
  File "C:\Users\bzdli\Anaconda3\envs\drl\lib\ntpath.py", line 204, in split
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType

I's using windows 10 and python 3.6.  Can someone figure this out?

fcampos, 2018-05-04:
I have also faced this issue... Has anyone found a solution yet?  Many thanks!

fcampos, 2018-05-04:
Ok, I have noticed there is a fix in pull request 'Load shared library properly on windows.'

Pizzafarmer, 2018-05-24:
Has anyone managed to fix this?

zovathio, 2018-06-22:
@bzdlity @fcampos Do you have any solution to fix the problem?
I also use Win10 and python 3.6.

alimuldal, 2018-10-10:
I believe this was fixed in ebc6442646c711ec5b908b6fc32befaea48da7fb - please comment if the error still occurs.

closed 2018-10-10 by alimuldal