google-deepmind/dm_control #30: Model used to train the agent

rohit497, 2018-03-27:
I'm trying to train a custom agent to learn how to swim in one of the environments provided by this repo using deep Q learning. However, I'm not sure how to get around the fact that the action space is continuous and the observation space is a dictionary of arrays, since all the deep Q models I've looked at initialize their `num_actions` and `num_states` variables as follows:

`num_actions = env.action_space.shape[0]
num_states = env.observation_space.shape[0]`

Is there some other way to train this model using `action_spec()` and `observation_spec()`? If so, are there examples that I can look at?

Thanks!

lisayan, 2018-03-28:
Also, for something simple like fish swimming to the target, how many time steps were used in each episode to train, and how many episodes were used in the model?

alimuldal, 2018-04-05:
The Benchmarking section in our [tech report][1] contains some baseline results for three different RL agents (A3C, DPG and D4PG) and a description of the models and training regimes we used to generate them. I believe the policies shown in the YouTube video were trained using D4PG, which is described in more detail [here][2].

All control suite environments expect a float array of continuous actions, so in order to train DQN on these environments you will need to discretize the action space. For example you could write an environment wrapper that overrides the [`step` method][3] to convert the discrete output of your DQN agent to an array of continuous actions that can be passed to the wrapped environment. Similarly, if your agent requires a single observation array then you could write a wrapper that concatenates the arrays in the observation dict into a single array (for example, see [`flatten_observation`][4]).

[1]: https://arxiv.org/abs/1801.00690
[2]: https://openreview.net/pdf?id=SyZipzbCb
[3]: https://github.com/deepmind/dm_control/blob/master/dm_control/rl/control.py#L98
[4]: https://github.com/deepmind/dm_control/blob/master/dm_control/rl/control.py#L369

closed 2018-04-05 by alimuldal