google-deepmind/dm_control #32: NoneType' object has no attribute 'destroy_window'

Jin0932, 2018-04-13:
I use the code in README.md,  After I install mujoco and dm_control， when I run code ,it remind me th efollowing error:

Exception AttributeError: "'NoneType' object has no attribute 'destroy_window'" in <bound method GLFWContext.del of <dm_control.render.glfw_renderer.GLFWContext object at 0x7ffba109c110>> ignored

So I have to ask for your help ,

alimuldal, 2018-04-13:
That looks like a minor bug in `GLFWContext._free`. What's probably happening is that when your Python interpreter terminates, `glfw.destroy_window` is being garbage-collected before `GLFWContext.__del__` has been called, so the call to `glfw.destroy_window` in `GLFWContext._free` fails. This isn't an ideal situation, but exceptions in `__del__` methods are non-fatal (which is why the error message says "ignored"). Did any other errors occur, or did the example code otherwise work as expected? You should see it print out 1000 tuples that look something like this:
```python
(0.019122017258586969, 1.0, OrderedDict([('orientations', array([-0.45806725,  0.88891754,  0.85568349,  0.51749953, -0.40150487,
        0.91585689, -0.30560896,  0.95215711,  0.79778138,  0.60294682,
       -0.96328198,  0.26849177, -0.98554371,  0.16942137])), ('height', 0.30465686511298051), ('velocity', array([  0.35438793,   2.0158634 ,  -3.99206314,  -9.94227997,
         5.68466767,  30.8521611 ,  -9.7774301 ,   2.43226494,   0.7295889 ]))]))
```

Jin0932, 2018-04-14:
HI alimuldal,
  thank you for reading this problem,   this is complete information. Some of the previous data is omitted.

.......
OrderedDict([('orientations', array([-0.27023141,  0.9627954 ,  0.34712266,  0.93781974, -0.3317505 ,
        0.94336717, -0.92409071,  0.3821732 ,  0.29170452,  0.95650848,
       -0.5014389 ,  0.86519306, -0.78349546,  0.62139751])), ('height', 0.12070568657821856), ('velocity', array([  0.90254797,   0.07358524,   1.34696174,   1.42850794,
         1.48089243,   1.67261856,   6.83868214,  -5.21467023, -11.93879889]))]))
Exception AttributeError: "'NoneType' object has no attribute 'destroy_window'" in <bound method GLFWContext.__del__ of <dm_control.render.glfw_renderer.GLFWContext object at 0x7f908af20350>> ignored

Process finished with exit code 0

Running Environment
Ubuntu16.04LTS,
python2.7
mujoco150

so I have to use the following code, But I think dp_control not need do nothing such as plot window , except care about  the algorithm
from dm_control import suite
import matplotlib.pyplot as plt
import numpy as np

**max_frame = 90

width = 480
height = 480
video = np.zeros((90, height, 2 * width, 3), dtype=np.uint8)**

# Load one task:
env = suite.load(domain_name="cartpole", task_name="swingup")

# Step through an episode and print out reward, discount and observation.
action_spec = env.action_spec()
time_step = env.reset()
while not time_step.last():
  for i in range(max_frame):
    action = np.random.uniform(action_spec.minimum,
                             action_spec.maximum,
                             size=action_spec.shape)
    time_step = env.step(action)
    **video[i] = np.hstack([env.physics.render(height, width, camera_id=0),
                          env.physics.render(height, width, camera_id=1)])**
    #print(time_step.reward, time_step.discount, time_step.observation)
  for i in range(max_frame):
    **img = plt.imshow(video[i])
    plt.pause(0.01)  # Need min display time > 0.0.
    plt.draw()**

I have try to use different lib(libglew, libglfw), but all failure,so what I do?

thanks

alimuldal, 2018-10-10:
We just pushed several updates to the MuJoCo bindings rendering stack which mean that the particular error mentioned above will not occur any more. You should still expect to see messages of the form
```
Exception RuntimeError: RuntimeError('This executor has already been terminated.',) in <function callback at 0x7f50315e8b90> ignored
```
when the Python interpreter process terminates (even when it terminates successfully). Although they are a bit annoying, these warnings can be safely ignored.

closed 2018-10-10 by alimuldal