google-deepmind/dm_control #39: Rendering and Saving

rajcscw, 2018-08-21:
Is there an easy way of rendering and saving the screens as videos like a wrapper in gym environments? 
`env = SomeWrapper(save_loc="path")` and `env.render()`

rajcscw, 2018-08-22:
That's a neat solution. Thanks for the code example!

wassimseif, 2018-09-27:
Hey doesn't the control suite have a similar renderer like openai gym ?

wpumacay, 2018-09-27:
Hi @wassimseif .

I guess not. Under the hood, mujoco has its own visualizer built with opengl and some libraries like glfw and glew. The renderer module here wraps some of the functionality and allows you to interact with that visualizer by requesting the rendered frame from the visualizer, but is not interactive i.e. it doesn't create a window where to render the frame. As far as I've checked, gym does something similar, but it creates a window to render the frame ( they don't have a separate renderer either, but use the internal visualizer. A bit more interactively, though ).

Melanol, 2022-07-14:
@JAEarly Saving video works, but while playing, I get a lot of these errors:
QObject::moveToThread: Current thread (0x55c6e4326da0) is not the object's thread (0x55c6e41813d0).
Cannot move to target thread (0x55c6e4326da0)

And then it ends with this (although, it manages to play the entire video):
Traceback (most recent call last):
  File "/home/mel/PycharmProjects/dm_control_tripod/render.py", line 57, in <module>
    play()
  File "/home/mel/PycharmProjects/dm_control_tripod/render.py", line 46, in play
    cv2.imshow('Playback', frame)
cv2.error: OpenCV(4.6.0) /io/opencv/modules/highgui/src/window.cpp:967: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'imshow'