google-deepmind/dm_control #48: GLEW initalization error ,it comes again

CharlieLuuke, 2018-10-12:
I updated dm_control(with viewer...) today.But when I run sample code in Quickstart, the error comes again.
>>> from dm_control import suite
>>> import numpy as np
>>> 
>>> # Load one task:
... env = suite.load(domain_name="cartpole", task_name="swingup")
>>> 
>>> # Iterate over a task set:
... for domain_name, task_name in suite.BENCHMARKING:
...   env = suite.load(domain_name, task_name)
... 
WARNING: Logging before flag parsing goes to stderr.
F1012 12:47:55.453529 140583399446272 core.py:93] **GLEW initalization error: Missing GL version**

However, it works well when I run sample code in "Interactive environment viewer".
I've reinstalled old "dm_control"package，the Quickstart run well(0 error,0 warning). 
I'm so confused why the error comes again.（Write the issue for the first time, please enlighten me.）
(ubantu 16.04)

alimuldal, 2018-10-12:
Reopening because I'm able to reproduce this error. It only seems to occur when using the GLFW backend (could you try setting the environment variable `MUJOCO_GL=osmesa` before launching the interpreter?). I suspect there may be a race condition involving context cleanup.

CharlieLuuke, 2018-10-13:
After I added the defensive call, the program works fine. ：）
I think this is just a matter of expediency. I agree that this problem stems from the call to GLFW backend, but this is where I am curious. I don't know how to force the program to use osmesa (failed), I will continue to follow.

naeioi, 2018-10-16:
we are still experiencing this issue with https://github.com/deepmind/dm_control/commit/0277e4352c57287c9f8497cb600a2bbe6d82ac45 and Mujoco v1.50.

alimuldal, 2018-10-16:
@naeioi  Could you try adding a defensive call to `env.physics.contexts.mujoco.free()` (as in fe1a361), or else using OSMesa instead of GLFW (`MUJOCO_GL=osmesa`)?

naeioi, 2018-10-17:
@alimuldal Thanks for advice. I tried calling `env.physics.contexts.mujoco.free()` every time after finishing using `dm_control.suite` but with no luck this exception still occurs nondeterministically (We are running against all available dm_control environments in CI). So we decided to use last stable version https://github.com/deepmind/dm_control/commit/c24ec9f5f3cb3c25c6571c89c9f60bf3350f5711 for now.

commit 64e594f3 (2018-10-19): Fix a bug in PassthroughRenderExecutor that causes a race condition when repeatedly instantiating and destroying rendering contexts.

Fixes: #48
PiperOrigin-RevId: 217892325
closed 2018-10-19 by paksas