google-deepmind/dm_control #53: New domain and task

michtesar, 2018-10-24:
Suppose I would like to write my own domain and task.

I copied humanoid.xml and humanoid.py, renamed to the same name and also in all occurrence in those files. I included __init__.py from suite and added import for the new suit I just copied.

Since there is no proper documentation I assume making this as playground is the best, right? I know I can register Gym environments in __init__.py very easily. Is there any way how to develop my own domain and task? Thank you!

alimuldal, 2018-10-26:
You would need to create a [`containers.TaggedTasks()`][tagged_tasks] instance in your new module (called `SUITE` by default), and then for each new task that you want to add, wrap the function that returns the environment using the `@SUITE.add` decorator (optional: provide some string tags).

For example:
```python
from dm_control.utils import containers

SUITE = containers.TaggedTasks()

@SUITE.add('some_tag', 'some_other_tag')
def my_new_task(time_limit=_DEFAULT_TIME_LIMIT, random=None,
                              environment_kwargs=None):
   ...
   return control.Environment(...)
```
Then in `dm_control/suite/__init__.py` you just need to import the module containing your new tasks. The tasks will then be loadable using `suite.load(domain='my_module_name', task='my_new_task', ...)`.

[tagged_tasks]: https://github.com/deepmind/dm_control/blob/master/dm_control/utils/containers.py#L25

closed 2018-10-26 by alimuldal

aaronsnoswell, 2019-05-07:
(For future readers): also see some of the comments on #46.

jinfagang, 2022-04-28:
How to specific xml file of my model here?