google-deepmind/dm_control #54: How to add new objects in one domain or environment?

WuXinyang2012, 2018-10-30:
For example, in the domain Point-mass, which is introduced in your thesis "DeepMind Control Suite", I want to add a new collision point near the target, which works like a trap, every time when the moving point touches this collision point, it will get a negative reward -1. 

As a result, I can finally train an agent, who can not only reach the desired target but also avoid the possible collisions when planning the path. The idea is like in a factory, there are not only robots but also human workers, and when the robot do some tasks, it shall avoid the collision and hurting humans.

I tried to modify the XML file to add the point, but it appears as only a static point in the GUI and the agent just ignores it, because I do not know how to specify a negative reward for this point. And btw, I want to make this point dynamic, just like a walking human.

I would be really grateful if you can offer some documentation or tutorials about how to create new objects in an existing environment or totally creating a new environment.

DanielTakeshi, 2019-02-23:
A related issue: https://github.com/deepmind/dm_control/issues/46 which is asking on documentation to create a new environment.

JeanKaddour, 2019-05-29:
Similar to OP's question; is there a way to change the parameters of an environment or the underlying mujoco model on the fly, i.e. without changing values in the XML file? Any help would be massively appreciated.

wpumacay, 2019-05-29:
Hi @JeanKaddour. Do you mean changing properties like stiffness, damping, sizes, etc. at runtime?. Perhaps you could directly access the internal structures of a mujoco model by using something like:

* physics.data.[field_you_want] = [something]
* physics.model.[field_you_want] = [something]

With mujoco's C/C++ API you can access directly these properties using the mjModel and mjData data structures and change them as you need, although I think I saw in a post something related to some fields requiring a reset (I'm not sure, I think Emo was talking about meshes, which needed to be updated for GPU, or it was masses). You can check mujoco's forum for this as well to double check. I've played around with some properties like sizes, rbound, meshes, but there are some I haven't tested, so I'm not sure which ones you're allowed to change (although I'm pretty sure most of them can be changed).

If you want to spawn things at runtime I think that's still not possible with the current version of Mujoco (as they mentioned in the forums, this is a feature for a future release). Models from XML pass through the Mujoco compiler, and then they remain "static" in the sense that you cannot add or delete parts of the model as you please. Still, if you want to make something like having some extra objects that you want to move and change its properties, I think you should wrap them with a class of your own, change its properties in that class and then request the changes in the simulation using its own id (obj_id, geom_id). You would need to have these in your xml, but you could just use them as you need by hiding them momentarily. Perhaps Controlsuite's composer API could help you with this, by creating these objects in the XML (injecting them) with appropriate names, save these names in your wrapper for later usage and implement the logic you need through changes to physics.data and physics.model.

Hope this helps.

guhur, 2019-07-09:
Spawing new objects seem to be possible with `mjv_addGeoms`. Have anyone already used it in this wrapper?

alimuldal, 2019-07-12:
@guhur That function only adds geoms to a scene for visualization purposes  (functions whose names start with `mjv_` all relate to visualisation). It doesn't affect the underlying physics simulation. You can tell that from [its signature](http://www.mujoco.org/book/APIreference.html#mjv_addGeoms):
```cpp
void mjv_addGeoms(const mjModel* m, mjData* d, const mjvOption* opt,
                  const mjvPerturb* pert, int catmask, mjvScene* scn);
```
Note that `mjModel*` and `mjData*` are both `const`, so it's not modifying the physics simulation.

There's no way to dynamically add or remove things from a MuJoCo simulation - you have to recompile the model from the XML. There are various hacky workarounds that people sometime use in order to avoid recompilation. For example you can define extra bodies/geoms etc. in the XML and "hide" them (e.g. by placing them really far away from the rest of the model). Then later on when they are needed you can move them into the scene.

closed 2020-06-23 by alimuldal