google-deepmind/dm_control #57: Can't import viewer on Arch Linux

KeAWang, 2018-10-31:
Every time I try to `from dm_control import viewer`, Python throws the follow error
```
>>> from dm_control import viewer
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "PYTHON_INSTALLATION/lib/python3.7/site-packages/dm_control/viewer/__init__.py", line 22, in <module>
    from dm_control.viewer import application
  File "PYTHON_INSTALLATION/lib/python3.7/site-packages/dm_control/viewer/application.py", line 23, in <module>
    from dm_control import render
  File "PYTHON_INSTALLATION/lib/python3.7/site-packages/dm_control/render/__init__.py", line 62, in <module>
    Renderer = import_func()
  File "PYTHON_INSTALLATION/lib/python3.7/site-packages/dm_control/render/__init__.py", line 38, in _import_osmesa
    from dm_control.render.pyopengl.osmesa_renderer import OSMesaContext
  File "PYTHON_INSTALLATION/lib/python3.7/site-packages/dm_control/render/pyopengl/osmesa_renderer.py", line 41, in <module>
    from OpenGL import GL
  File "PYTHON_INSTALLATION/lib/python3.7/site-packages/OpenGL/GL/__init__.py", line 4, in <module>
    from OpenGL.GL.VERSION.GL_1_1 import *
  File "PYTHON_INSTALLATION/lib/python3.7/site-packages/OpenGL/GL/VERSION/GL_1_1.py", line 14, in <module>
    from OpenGL.raw.GL.VERSION.GL_1_1 import *
  File "PYTHON_INSTALLATION/lib/python3.7/site-packages/OpenGL/raw/GL/VERSION/GL_1_1.py", line 7, in <module>
    from OpenGL.raw.GL import _errors
  File "PYTHON_INSTALLATION/lib/python3.7/site-packages/OpenGL/raw/GL/_errors.py", line 4, in <module>
    _error_checker = _ErrorChecker( _p, _p.GL.glGetError )
  File "PYTHON_INSTALLATION/lib/python3.7/ctypes/__init__.py", line 369, in __getattr__
    func = self.__getitem__(name)
  File "PYTHON_INSTALLATION/lib/python3.7/ctypes/__init__.py", line 374, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /usr/lib/libOSMesa.so.8: undefined symbol: glGetError
```
which seems to be caused by PyOpenGL's incompatibility with mesa-18.2.3-1 on Arch Linux. I've open an issue on PyOpenGL here: https://github.com/mcfletch/pyopengl/issues/18
but I'm wondering if anyone else has run into this issue. If so, how did you get around it?

alimuldal, 2018-11-01:
See https://github.com/deepmind/dm_control/issues/55#issuecomment-434503122. OSMesa is a non-starter as far as the viewer is concerned, since it requires GLFW rather than OSMesa  (OSMesa is only supported for offscreen rendering on Linux).

By default the `render` module tries to import the GLFW backend first, then falls back on OSMesa if GLFW is not available. Some possible explanations for this are missing shared libraries or X11 not being available. What happens if you try to force the GLFW backend by setting `MUJOCO_GL=glfw` when launching your interpreter?

KeAWang, 2018-11-01:
Awesome, I didn't know that `render` first tries to import GLFW. I installed GLFW3 and it works fine now without having to change `MUJOCO_GL`. Thanks so much!

closed 2018-11-01 by KeAWang