google-deepmind/dm_control #58: Segfault when calling from_xml_path with meshes files

tkelestemur, 2018-10-31:
Duplication of #11 but I wanted to open a new issue with my .xml. My setup is Ubuntu 16.04, MuJoCo 2.0 and Python 2.7. The .xml file works fine with `simulate` example of MuJoCo Pro. Currently, I am able to use call from_binary_path but it is not very convenient since everytime I want to change something in the environment I have to `compile` the binary model from xml.

Here is my [.xml file](https://gist.github.com/tkelestemur/25b57cdb5009d7f24ed1ffd3334f122f). I can share my meshes files as well if you want to reproduce.

alimuldal, 2018-11-01:
Could you share the meshes as well, please?

tkelestemur, 2018-11-01:
[Here](https://github.com/tkelestemur/sawyer_mujoco/tree/master/meshes) they are.

alimuldal, 2018-11-01:
Hmm, I'm not able to reproduce the segfault. I cloned your repository, edited `sawyer_mujoco.xml` to set  `meshdir="../meshes/"`, and uncommented the lines at the end of `simulate.py` that launches the viewer with a random policy. `python simulate.py` works perfectly for me:
[image]

Could you paste the full traceback?

alimuldal, 2018-11-01:
I think this is related to the ABI conflict mentioned in https://github.com/deepmind/dm_control/issues/24#issuecomment-433733298. I was able to reproduce the segfault in a clone of your repository by setting `meshdir="../meshes"` in `sawyer_mujoco.xml` and changing `simulate.py` to load from the XML path instead of the binary. However, if I launch using `LD_PRELOAD=libstdc++.6.so python simulate.py` it works correctly. Again, the problem seems to be specific to the GLFW backend, since `MUJOCO_GL=osmesa  python simulate.py` also doesn't segfault (although with OSMesa you won't be able to use the viewer).

alimuldal, 2018-11-12:
Fixed in e374a4a75461d07a80a1316463c45f9ccbb45324

closed 2018-11-12 by alimuldal