google-deepmind/dm_control #59: Does this control suite support collision detection and contact force?

WuXinyang2012, 2018-11-09:
By looking through the mujoco documentation, I found that the mujoco supports collision detection and all the active contacts are stored in the variable mjData.contact, by which man can also calculate the contact force.
Does this control suite have any interface to get the collision and contact information? How can I access the mjData.contact variable at runtime with this suite? 

Thanks in advance.

alimuldal, 2018-11-12:
Active contacts are accessible via `physics.data.contact`. This is a 1D [`numpy.recarray`][recarray] of shape `(physics.data.ncon,)` that exposes all the fields in [`mjContact`][mjContact] struct for all of the currently active contacts. For example, you can get the geom IDs for each active contact pair via `physics.data.contact.geom1` and `physics.data.contact.geom2`.

The recommended way to get the contact force for a particular contact is to use [`mj_contactForce`][mj_contactForce], which you can access via `dm_control.mujoco.wrapper.mjbindings.mjlib.mj_contactForce`. Depending on your use-case you might also be able to use a touch sensor (see [`manipulator.xml`][manipulator] domain for example).

[recarray]: https://docs.scipy.org/doc/numpy-1.15.1/reference/generated/numpy.recarray.html
[mjContact]: http://www.mujoco.org/book/reference.html#mjContact
[mj_contactForce]: http://www.mujoco.org/book/reference.html#mj_contactForce
[manipulator]: https://github.com/deepmind/dm_control/blob/master/dm_control/suite/manipulator.xml#L195-L201

closed 2018-11-12 by alimuldal

wanxinjin, 2022-05-15:
Thanks for the explanation. A follow-up question:

In my case, the `sim.data.contact` has a shape of `sim.model.nconmax`, which is far larger than `sim.data.ncon`. 
I assume those additional ones are inactive contacts. How do i tell which are the `sim.data.ncon` contacts and which are not? Do the active contacts always come first, i.e., `sim.data.contact[0: sim.data.ncon]` ?

Thanks

yuvaltassa, 2022-05-15:
Yes, that is correct. 

Similarly, quantities which are pre-allocated with `model.njmax` are only populated up to `data.nefc`.