google-deepmind/dm_control #61: Runtime error with a simple MJCF example

Xingyu-Lin, 2018-11-12:
Hi, I tried to run the following simple MJCF model from the tech report:

```
import numpy as np
from dm_control import mujoco
simple_MJCF = """
<mujoco>
<worldbody>
<light name ="top" pos ="0 0 1.5"/>
<geom name ="floor" type ="plane" size ="1 1 .1"/>
<body name ="box" pos ="0 0 .3" >
<joint name ="up_down" type ="slide" axis ="0 0 1"/>
<geom name ="box" type ="box" size =".2 .2 .2" rgba ="1 0 0 1"/>
<geom name ="sphere" pos =".2 .2 .2" size =".1" rgba ="0 1 0 1"/>
</body>
</worldbody>
</mujoco>
"""
physics = mujoco.Physics.from_xml_string(simple_MJCF)
pixels = physics.render()
```
The code terminates with the following message:

Exception ignored in: <function _create_finalizer.<locals>.callback at 0x7f087dea4510>
Traceback (most recent call last):
  File "/home/xingyu/software/miniconda3/envs/dm_control/lib/python3.6/site-packages/dm_control/mujoco/wrapper/core.py", line 305, in callback
  File "/home/xingyu/software/miniconda3/envs/dm_control/lib/python3.6/site-packages/dm_control/mujoco/wrapper/core.py", line 788, in finalize_mjr_context
  File "/home/xingyu/software/miniconda3/envs/dm_control/lib/python3.6/contextlib.py", line 81, in __enter__
  File "/home/xingyu/software/miniconda3/envs/dm_control/lib/python3.6/site-packages/dm_control/render/base.py", line 109, in make_current
  File "/home/xingyu/software/miniconda3/envs/dm_control/lib/python3.6/contextlib.py", line 81, in __enter__
  File "/home/xingyu/software/miniconda3/envs/dm_control/lib/python3.6/site-packages/dm_control/render/executor/render_executor.py", line 95, in execution_context
  File "/home/xingyu/software/miniconda3/envs/dm_control/lib/python3.6/site-packages/dm_control/render/executor/render_executor.py", line 90, in _check_not_terminated
RuntimeError: This executor has already been terminated.

Any idea on how to solve this?

alimuldal, 2018-11-12:
It's normal to see output like that when the interpreter process terminates (see https://github.com/deepmind/dm_control/issues/32#issuecomment-428725287). They are a bit annoying, but otherwise benign. Do any other exceptions occur further up in the traceback, or is it just the `RuntimeError` shown above?

Xingyu-Lin, 2018-11-12:
Thanks! This issue is solved!

closed 2018-11-12 by alimuldal

fabawi, 2018-12-07:
This problem doesn't appear to be solved. The executor terminates whenever I try to load an mjcf script. The problem does not occur when loading environments from the suite

FreescaleFlyaway, 2019-04-21:
hello~ I wanna know whether you have fixed the problem or not?
I get a similar error, could u please help me?

alimuldal, 2019-04-21:
Please see my other comments. You should expect to see that `RuntimeError` when the interpreter process terminates - if that's the only error you're seeing then you can safely ignore it.

FreescaleFlyaway, 2019-04-24:
> Please see my other comments. You should expect to see that `RuntimeError` when the interpreter process terminates - if that's the only error you're seeing then you can safely ignore it.

you mean that it is a normal situation or kind of no error if "RuntimeError" is the only error, right? 
I get it. Thank you!