google-deepmind/dm_control #62: Can not use along with matplotlib

Xingyu-Lin, 2018-11-12:
Loading MJCF and visualize with matplotlib will crash (MacOS Mojave 10.14.1). Reproduce with the following code:
```
from matplotlib import pyplot as plt
import numpy as np
from dm_control import mujoco
simple_MJCF = """
<mujoco>
<worldbody>
<light name ="top" pos ="0 0 1.5"/>
<geom name ="floor" type ="plane" size ="1 1 .1"/>
<body name ="box" pos ="0 0 .3" >
<joint name ="up_down" type ="slide" axis ="0 0 1"/>
<geom name ="box" type ="box" size =".2 .2 .2" rgba ="1 0 0 1"/>
<geom name ="sphere" pos =".2 .2 .2" size =".1" rgba ="0 1 0 1"/>
</body>
</worldbody>
</mujoco>
"""
physics = mujoco.Physics.from_xml_string(simple_MJCF)
pixels = physics.render()
plt.imshow(pixels)
plt.savefig('./test.png')
```

The error message is

> 2018-11-12 16:13:15.266 python[37560:1214767] -[GLFWApplication _setup:]: unrecognized selector sent to instance 0x7f7f80990c00
> 2018-11-12 16:13:15.266 python[37560:1214767] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[GLFWApplication _setup:]: unrecognized selector sent to instance 0x7f7f80990c00'
> *** First throw call stack:
> (
> 	0   CoreFoundation                      0x00007fff2ee07e65 __exceptionPreprocess + 256
> 	1   libobjc.A.dylib                     0x00007fff5ae5f720 objc_exception_throw + 48
> 	2   CoreFoundation                      0x00007fff2ee8522d -[NSObject(NSObject) __retain_OA] + 0
> 	3   CoreFoundation                      0x00007fff2eda9820 ___forwarding___ + 1486
> 	4   CoreFoundation                      0x00007fff2eda91c8 _CF_forwarding_prep_0 + 120
> 	5   libtk8.6.dylib                      0x000000011b3e246f TkpInit + 383
> 	6   libtk8.6.dylib                      0x000000011b33b21e Initialize + 2622
> 	7   _tkinter.cpython-36m-darwin.so      0x000000011b161a16 _tkinter_create + 1174
> 	8   python                              0x000000010e167ae8 _PyCFunction_FastCallDict + 200
> 	9   python                              0x000000010e23f3e2 call_function + 146
> 	10  python                              0x000000010e23d0c8 _PyEval_EvalFrameDefault + 46936
> 	11  python                              0x000000010e23073b _PyEval_EvalCodeWithName + 427
> 	12  python                              0x000000010e23ff83 _PyFunction_FastCallDict + 355
> 	13  python                              0x000000010e0e5b00 _PyObject_FastCallDict + 320
> 	14  python                              0x000000010e10cf68 method_call + 136
> 	15  python                              0x000000010e0ed13e PyObject_Call + 62
> 	16  python                              0x000000010e18f485 slot_tp_init + 117
> 	17  python                              0x000000010e193981 type_call + 241
> 	18  python                              0x000000010e0e5a71 _PyObject_FastCallDict + 177
> 	19  python                              0x000000010e0edca7 _PyObject_FastCallKeywords + 327
> 	20  python                              0x000000010e23f492 call_function + 322
> 	21  python                              0x000000010e23d178 _PyEval_EvalFrameDefault + 47112
> 	22  python                              0x000000010e23f79f fast_function + 191
> 	23  python                              0x000000010e23f42e call_function + 222
> 	24  python                              0x000000010e23d0c8 _PyEval_EvalFrameDefault + 46936
> 	25  python                              0x000000010e23073b _PyEval_EvalCodeWithName + 427
> 	26  python                              0x000000010e23ff83 _PyFunction_FastCallDict + 355
> 	27  python                              0x000000010e0e5b00 _PyObject_FastCallDict + 320
> 	28  python                              0x000000010e10cf68 method_call + 136
> 	29  python                              0x000000010e0ed13e PyObject_Call + 62
> 	30  python                              0x000000010e23d2be _PyEval_EvalFrameDefault + 47438
> 	31  python                              0x000000010e23073b _PyEval_EvalCodeWithName + 427
> 	32  python                              0x000000010e23f83c fast_function + 348
> 	33  python                              0x000000010e23f42e call_function + 222
> 	34  python                              0x000000010e23d0c8 _PyEval_EvalFrameDefault + 46936
> 	35  python                              0x000000010e23f79f fast_function + 191
> 	36  python                              0x000000010e23f42e call_function + 222
> 	37  python                              0x000000010e23d0c8 _PyEval_EvalFrameDefault + 46936
> 	38  python                              0x000000010e23073b _PyEval_EvalCodeWithName + 427
> 	39  python                              0x000000010e23f83c fast_function + 348
> 	40  python                              0x000000010e23f42e call_function + 222
> 	41  python                              0x000000010e23d0c8 _PyEval_EvalFrameDefault + 46936
> 	42  python                              0x000000010e23073b _PyEval_EvalCodeWithName + 427
> 	43  python                              0x000000010e23f83c fast_function + 348
> 	44  python                              0x000000010e23f42e call_function + 222
> 	45  python                              0x000000010e23d0c8 _PyEval_EvalFrameDefault + 46936
> 	46  python                              0x000000010e23073b _PyEval_EvalCodeWithName + 427
> 	47  python                              0x000000010e289c0c PyRun_FileExFlags + 252
> 	48  python                              0x000000010e2893be PyRun_SimpleFileExFlags + 366
> 	49  python                              0x000000010e2af216 Py_Main + 3718
> 	50  python                              0x000000010e0ddb0d main + 509
> 	51  libdyld.dylib                       0x00007fff5bf2e08d start + 1
> )
> libc++abi.dylib: terminating with uncaught exception of type NSException
> Abort trap: 6

The issue seems very similar to this one: https://github.com/deepmind/pysc2/issues/2

saran-t, 2018-11-13:
@Xingyu-Lin Could you elaborate on how you installed things? I'm currently only able to test on macOS 10.13.6 (High Sierra), with Python 3.6 and GLFW installed through pip as documented in the README.

In my setup, if I do `from matplotlib import pyplot as plt` as you did in your snippet, I get the following error message:

```ImportError: Python is not installed as a framework. The Mac OS X backend will not be able to function correctly if Python is not installed as a framework. See the Python documentation for more information on installing Python as a framework on Mac OS X. Please either reinstall Python as a framework, or try one of the other backends. If you are using (Ana)Conda please install python.app and replace the use of 'python' with 'pythonw'. See 'Working with Matplotlib on OSX' in the Matplotlib FAQ for more information.```

However, if I import matplotlib *after* I've created the `Physics` object, everything seems to work fine. (Note that I have not tried to install Python as a framework as requested by matplotlib. That might also make a difference.)

alimuldal, 2019-02-25:
Closing since we don't have enough information to diagnose or reproduce this

closed 2019-02-25 by alimuldal

camall3n, 2020-09-21:
I'm seeing the same crash with the following program:
```python
import matplotlib.pyplot as plt
from dm_control import suite

plt.plot()
plt.show()
```

If I disable the dm_control import, the empty figure displays normally. If I leave it in, I get the following error:

```
2020-09-21 09:26:23.494 Python[24974:1597019] -[NSApplication _setup:]: unrecognized selector sent to instance 0x7fa5e2974e80
2020-09-21 09:26:23.496 Python[24974:1597019] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplication _setup:]: unrecognized selector sent to instance 0x7fa5e2974e80'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff344e1acd __exceptionPreprocess + 256
	1   libobjc.A.dylib                     0x00007fff5ebe5a17 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff3455b8d6 -[NSObject(NSObject) __retain_OA] + 0
	3   CoreFoundation                      0x00007fff3448393f ___forwarding___ + 1485
	4   CoreFoundation                      0x00007fff344832e8 _CF_forwarding_prep_0 + 120
	5   Tk                                  0x00007fff4026e7d8 TkpInit + 466
	6   Tk                                  0x00007fff401edb86 Tk_Init + 1706
[...]
	54  Python                              0x000000010b315d0b _PyEval_EvalFrameDefault + 6576
	55  Python                              0x000000010b31d7a6 _PyEval_EvalCodeWithName + 1870
	56  Python                              0x000000010b2876e5 _PyFunction_FastCallDict + 441
	57  Python                              0x000000010b35a35c pymain_run_module + 147
	58  Python                              0x000000010b3593f5 pymain_main + 4196
	59  Python                              0x000000010b359f80 _Py_UnixMain + 56
	60  libdyld.dylib                       0x00007fff603b43d5 start + 1
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
```

I'm running Python 3.7.4 on MacOS 10.14.6 and `pip list` shows the following versions:
```
dm-control             0.0.322773188
dm-env                 1.2
dm-tree                0.1.5
matplotlib             3.3.2
```