google-deepmind/dm_control #63: Headless GPU Rendering

amandlek, 2018-11-27:
I'm wondering whether anyone has gotten headless GPU rendering to work. On our Ubuntu machine with a display, the following [script](https://www.dropbox.com/s/56p9wftglyt4dkw/test_dmcontrol.py?dl=0) is able to use the GPU for rendering images, but when running the script headless (i.e. through ssh) we are unable to get it to work with the GPU.

alimuldal, 2018-11-28:
The GLFW hardware rendering backend requires X11, so it won't work on a headless machine. At the moment your best option is to use software rendering on headless machines, but we are also working on an EGL backend which will allow for hardware rendering on headless machines.

amandlek, 2018-11-28:
When can we expect the EGL backend to be ready?

alimuldal, 2018-11-29:
I can't give you an exact timeline, but probably after NeurIPS

saran-t, 2018-11-30:
I've committed our initial version of headless EGL rendering code now. If you run dm_control on a Linux machine with an Nvidia GPU installed, this should now be the new default backend.

Technical caveat: we normally use "render executors" to ensure that all OpenGL calls occur on a separate, dedicated thread. However, this is currently not working with our open-sourced EGL code. This should not affect you unless you use multiple threads in your program. (If you do, you are responsible for making sure that the `Physics` object is created on the same thread as where `physics.render` is called.)

I have tested this on my own laptop with a GeForce 1050 GTX and it seems to work fine. However, I'd appreciate if people here could also test it out and reply here to say whether it is working for them or not.

If this is causing trouble for you, you can revert to software rendering by setting the environment variable `MUJOCO_GL=osmesa`.

closed 2018-11-30 by saran-t

amandlek, 2018-11-30:
@saran-t Works like a charm on our linux machine. Thanks for the help!

KeAWang, 2019-01-09:
EGL rendering seems to be causing segfaults on my system. Has anyone else had this issue https://github.com/deepmind/dm_control/issues/70?

DanielTakeshi, 2019-11-22:
Is EGL considered to be deprecated in favor of OSMesa?

commit 7c1f1070 (2018-11-30): Add support for Nvidia hardware-accelerated headless rendering via EGL.

After this change, headless EGL is used by default on machines with an Nvidia GPU installed.

Closes: #63
PiperOrigin-RevId: 223498983