google-deepmind/dm_control #64: Are dm_control environments deterministic?

amandlek, 2018-11-27:
I'm wondering if dm_control environments are deterministic. From running the script [here](https://www.dropbox.com/s/dbz8qev8mpm79dv/test_dm_playback.py?dl=0) it seems as though it is not, but I'm pretty sure MuJoCo is supposed to be deterministic. Note that you might need to wait for up to 100 iterations to see an assertion error.

yuvaltassa, 2018-11-27:
`dm_control` environments are deterministic. The caveat is that the fields touched by `get_state()` and `set_state()` do not _exactly_ correspond to the binary state of the simulator. The culprit is `physics.data.qacc_warmstart`. This is a field used by MuJoCo to accelerate the computations of the constraint solver. The effect on the "true" state (positions & velocities) is tiny, but will accumulate over time, as you have noticed.

To achieve exact deterministic rollouts, either set `qacc_warmstart[:] = 0` or call `physics.reset()` before setting the initial state of the two trajectories you are comparing. You could of course also save `qacc_warmstart` as a state variable, but that is not recommended. Yet another option is to disable the environment's [warmstart flag](http://mujoco.org/book/XMLreference.html#option-flag).

That said, _training_ with deterministic dynamics is not recommended, as policies will exploit the determinism to find semi-stable limit cycles, as beautifully illustrated by [Ben Recht and co](http://www.argmin.net/2018/03/20/mujocoloco/). We therefore strongly recommend using the [Gaussian Noise wrapper](https://github.com/deepmind/dm_control/commit/7511ba9ac4696b158c30cc562e7a969b1d714469) during training.

closed 2018-11-27 by yuvaltassa

amandlek, 2018-11-27:
Hi @yuvaltassa, I tried modifying the script to disable the warmstart - you can find it [here](https://www.dropbox.com/s/dbz8qev8mpm79dv/test_dm_playback.py?dl=0). It still fails from time to time - do you have any additional insight on what might be happening?

I completely agree on your point about training with deterministic dynamics - we would just like the capability to collect demonstrations and replay the actions deterministically to recover the original trajectory.

aaronsnoswell, 2019-05-07:
Hi @yuvaltassa. Thanks for this great library.

Can I suggest that [your comment above](https://github.com/deepmind/dm_control/issues/64#issuecomment-442041584) be promoted to a wiki page/section in the README somewhere? This is very helpful information for someone using this library :)