google-deepmind/dm_control #66: Pre-allocated contact buffer is full.

huiwenzhang, 2018-11-29:
Hi, I am trying to reproduce the experiment from paper "Learning human behaviors from motion capture by adversarial imitation". I used the code from https://github.com/huiwenzhang/merel-mocap-gail. Bascially the code uses  the GAIL implementation from OpenAI baselines package. While the environment uses the MuJoCo with the dm_control bindings. When I run the experiment, at beginning, it is fine. But after running about 400 iterations, it throws the following error. The error may come from the your package or may from the GAIL algorithm. So I am not sure I posted the issue in the right place. But according to the error, it seems that I didn't release memory or some reason. If you guys have any idea,  let me know. The errors are pasted below:

```
********** Iteration 413 ************
Optimizing Policy...
sampling
WARNING: Logging before flag parsing goes to stderr.
W1128 18:13:48.737738 140367550666496 core.py:88] Pre-allocated contact buffer is full. Increase nconmax above 100. Time = 0.0000.
Traceback (most recent call last):
  File "/home/huiwen/paper/learn-from-video/merel-mocap-gail/baselines/baselines/gail/run_mujoco.py", line 268, in <module>
    main(args)
  File "/home/huiwen/paper/learn-from-video/merel-mocap-gail/baselines/baselines/gail/run_mujoco.py", line 125, in main
    args.obs_only
  File "/home/huiwen/paper/learn-from-video/merel-mocap-gail/baselines/baselines/gail/run_mujoco.py", line 181, in train
    obs_only=obs_only)
  File "/home/huiwen/paper/learn-from-video/merel-mocap-gail/baselines/baselines/gail/trpo_mpi.py", line 316, in learn
    seg = seg_gen.__next__()
  File "/home/huiwen/paper/learn-from-video/merel-mocap-gail/baselines/baselines/gail/trpo_mpi.py", line 132, in traj_segment_generator
    env.reset()
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/site-packages/dm_control/rl/control.py", line 90, in reset
    self._task.initialize_episode(self._physics)
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/site-packages/dm_control/suite/humanoid_CMU.py", line 140, in initialize_episode
    physics.after_reset()
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/site-packages/dm_control/mujoco/engine.py", line 231, in after_reset
    self.forward()
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/site-packages/dm_control/mujoco/engine.py", line 241, in forward
    mjlib.mj_forward(self.model.ptr, self.data.ptr)
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/contextlib.py", line 88, in __exit__
    next(self.gen)
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/site-packages/dm_control/mujoco/engine.py", line 253, in check_invalid_state
    _INVALID_PHYSICS_STATE.format(warning_names=', '.join(warning_names)))
dm_control.rl.control.PhysicsError: Physics state is invalid. Warning(s) raised: mjWARN_CONTACTFULL
```
Thanks

wpumacay, 2018-11-29:
Hi, I run into the same issue when trying to spawn many agents into the world. It seems mujoco pre-allocates some memory under the hood for the contacts, joints, and some other stuff. As suggested by the warning, you could increase nconmax. I did it in the mjcf xml like this:

```xml
<mujoco model="planar walker">
  <include file="visual.xml"/>
  <include file="skybox.xml"/>
  <include file="materials.xml"/>

  <!-- Some other configuration -->

  <!-- at same level -->
  <size njmax="8000" nconmax="4000"/>

  <!-- Some other configuration -->
```
I used that size of the preallocated buffers because I was spawning 5-10 humanoids at the same time. You could increase it as you need (just the nconmax seems to be your issue, in my case I needed njmax too because of the joints). Just be aware, as you increase the size of your pre-allocated buffer you might increase the RAM usage in your system (in my case grew from ~40MB for a single model to ~900MB for 10 models)

alimuldal, 2018-11-29:
The number of active contacts and scalar constraints may vary dynamically at runtime (e.g. if a lot of objects suddenly collide with one another), whereas MuJoCo's contact and constraint buffers have fixed maximum sizes. If you manage to fill the contact or constraint buffer then MuJoCo will start ignoring contacts or constraints that don't fit in the buffer, which usually leads to weird behaviour (e.g. objects suddenly passing through one another). That's why we raise a `PhysicsError` in this situation.

MuJoCo tries to pick sensible default buffer sizes, but in general there isn't a reliable way to predict the maximum number of constraints or contacts. Allocating buffers that are _guaranteed_ to be big enough in the worst-case scenario where all geoms are colliding with all other geoms would require a huge amount of memory that probably isn't needed in practice.

As @wpumacay says, your best option is to increase `nconmax` and/or `njmax` until the error goes away, keeping in mind that this will increase your memory usage (quadratically, I think).

See the MuJoCo docs for more info:
http://www.mujoco.org/book/programming.html#siDiagnostics
http://www.mujoco.org/book/XMLreference.html#size

closed 2018-11-29 by alimuldal

huiwenzhang, 2018-11-30:
@alimuldal Thanks. @wpumacay  I didn't find the default settings for ncomax parameter in the xml file. Can I just add extra `size` tags there as you declared? Which implies this setting will override the default setting somewhere.

wpumacay, 2018-11-30:
@huiwenzhang , yes, it should work fine. I didn't have those either in my xml, but in the mujoco model samples folder I could find one that had those settings. You could check in the **model** folder in your mujoco unzipped folder (_~/.mujoco/YOUR_VERSION_OF_MUJOCO/model/_). I think the arm26.xml has those settings (if you have mjpro 200). For mjpro150, I think the humanoid100.xml has those settings as well. 

Indeed, this will override the default settings. You could also check if this is working fine by running either the **simulate** or **basic** examples in the bin folder in your mujoco folder (before running your own experiments with dm_control). Just change your xml and run those examples with your modified model, and it should give you a warning if it still needs a bigger buffer.