google-deepmind/dm_control #72: mjcf.namespace is None

jirenz, 2019-01-24:
I am trying to use the composer API but hitting the issue where `dm_control.mjcf.element.RootElement.spec.namespace` is `None`

Example: 
```python
from dm_control import mjcf
from dm_control import composer

mjcf_str = """<mujoco model="parent">
  <worldbody>
    <body>
      <geom name="foo" type="box" pos="-0.2 0 0.3" size="0.5 0.3 0.1"/>
      <site name="attachment_site" pos="1. 2. 3." quat="1. 0. 0. 1."/>
    </body>
  </worldbody>
</mujoco>"""

mjcf_model = mjcf.from_xml_string(mjcf_str)

print('model.spec.namespace', mjcf_model.spec.namespace)
# >> model.spec.namespace None

entity = composer.ModelWrapperEntity(mjcf_model)

task = composer.NullTask(entity)
env = composer.Environment(task)

binding = env.physics.bind(mjcf_model)

i = 0
while True:
    env.step()
    print(binding.xpos)

    if i % 100 == 0:
        print('step: {}'.format(i))
    i += 1
```

This gives the following error:

```
Traceback (most recent call last):
  File "/Users/jirenz/anaconda/envs/surreal/lib/python3.5/site-packages/dm_control/mjcf/physics.py", line 640, in bind
    binding = self._bindings[cache_key]
KeyError: MJCF Element: <mujoco model="parent">...</mujoco>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "dm-scratch-composer.py", line 24, in <module>
    binding = env.physics.bind(mjcf_model)
  File "/Users/jirenz/anaconda/envs/surreal/lib/python3.5/site-packages/dm_control/mjcf/physics.py", line 643, in bind
    namespace, named_index = names_from_elements(mjcf_elements)
  File "/Users/jirenz/anaconda/envs/surreal/lib/python3.5/site-packages/dm_control/mjcf/physics.py", line 170, in names_from_elements
    namespace = _get_namespace(mjcf_elements)
  File "/Users/jirenz/anaconda/envs/surreal/lib/python3.5/site-packages/dm_control/mjcf/physics.py", line 674, in _get_namespace
    namespace = element.spec.namespace.split(constants.NAMESPACE_SEPARATOR)[0]
AttributeError: 'NoneType' object has no attribute 'split'
Exception ignored in: <function _create_finalizer.<locals>.callback at 0x1098736a8>
Traceback (most recent call last):
  File "/Users/jirenz/anaconda/envs/surreal/lib/python3.5/site-packages/dm_control/mujoco/wrapper/core.py", line 305, in callback
    free_func(temp_ptr)
  File "/Users/jirenz/anaconda/envs/surreal/lib/python3.5/site-packages/dm_control/mujoco/wrapper/core.py", line 788, in finalize_mjr_context
    with gl_context.make_current() as ctx:
  File "/Users/jirenz/anaconda/envs/surreal/lib/python3.5/contextlib.py", line 59, in __enter__
    return next(self.gen)
  File "/Users/jirenz/anaconda/envs/surreal/lib/python3.5/site-packages/dm_control/render/base.py", line 109, in make_current
    with self._render_executor.execution_context() as ctx:
  File "/Users/jirenz/anaconda/envs/surreal/lib/python3.5/contextlib.py", line 59, in __enter__
    return next(self.gen)
  File "/Users/jirenz/anaconda/envs/surreal/lib/python3.5/site-packages/dm_control/render/executor/render_executor.py", line 95, in execution_context
    self._check_not_terminated()
  File "/Users/jirenz/anaconda/envs/surreal/lib/python3.5/site-packages/dm_control/render/executor/render_executor.py", line 90, in _check_not_terminated
    raise RuntimeError(_ALREADY_TERMINATED)
RuntimeError: This executor has already been terminated.
```

saran-t, 2019-01-29:
The error message could definitely be improved here. The actual problem is that the `physics.bind` API is supposed to be used on a particular element, not the entire model itself. 

For example, if you are after the `xpos` of the box geom in the model as specified above, you'd do `physics.bind(mjcf_model.find('geom', 'foo')).xpos`.

Since you are doing this multiple times in a loop, a more efficient way would be to store a reference, e.g. `foo_geom = mjcf_model.find('geom', 'foo')`. Then you could just `physics.bind(foo_geom).xpos` from there on.

The `spec` attribute of a PyMJCF object is supposed to be effectively for internal use only.

closed 2019-01-29 by saran-t

ishank-juneja, 2023-02-23:
I guess another way of putting saran-t's answer is that mujoco_element should be an `mjcf.Element` but not a `mjcf.RootElement` since the latter is an entire model.

Also the error message remains the same unhelpful one as of Feb 2023.

yuvaltassa, 2023-02-23:
A PR with a better error message would be most welcome.