google-deepmind/dm_control #74: Seg fault on skin element?

jirenz, 2019-01-29:
I don't know if this is from `dm_control` or `MuJoCo`, but rendering with glfw on mac gives my segfault when I try to use `<skin/>`. 

Sample script

```
from dm_control import mjcf
from dm_control import composer
from dm_control import viewer
from dm_control import suite

width = 320
height = 240
mjcf_str = """
<mujoco model="parent">
    <asset>
    <texture builtin="checker" height="512" name="texplane" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2" type="2d" width="512"/>
    <material name="floorplane" reflectance="0.5" texrepeat="1 1" texture="texplane" texuniform="true"/>
    </asset>
    <worldbody>
        <geom type="box" pos ="0.5 0 0.3" size="0.5 0.02 0.1" rgba="0 1 0 1"/>
        <geom condim="3" material="floorplane" name="floor" pos="0 0 -0.7" size="3 3 .125" type="plane"/>
        <body>
            <composite type="grid" count="9 9 1" spacing="0.05" offset="0.5 0 0.5">
                <skin material="floorplane"/>
                <geom size=".02"/>
            </composite>
        </body>
    </worldbody>
</mujoco>"""
world = mjcf.from_xml_string(mjcf_str)
world_entity = composer.ModelWrapperEntity(world)
task = composer.NullTask(world_entity)
task.control_timestep = 0.02
env = composer.Environment(task)
viewer.launch(env)
```

jirenz, 2019-01-29:
```
Process:               python3.5 [1206]
Path:                  /Users/USER/*/python
Identifier:            python3.5
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        bash [1054]
Responsible:           python3.5 [1206]
User ID:               501

Date/Time:             2019-01-29 08:25:21.956 -0800
OS Version:            Mac OS X 10.14.2 (18C54)
Report Version:        12
Anonymous UUID:        4BBCA77E-4B43-88C6-CA7C-8ED2CFCD7B9F

Sleep/Wake UUID:       511D5D9C-0779-4C6B-9B74-C0099A8D1C55

Time Awake Since Boot: 1000 seconds
Time Since Wake:       550 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [1206]

VM Regions Near 0:
--> 
    __TEXT                 000000010e9c0000-000000010e9c2000 [    8K] r-x/rwx SM=COW  /Users/USER/*/*.5

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_platform.dylib      	0x00007fff6d163975 _platform_bzero$VARIANT$Haswell + 53
1   libmujoco200.dylib            	0x0000000113f24b4c mjv_updateSkin + 188
2   _ctypes.cpython-35m-darwin.so 	0x000000010fe54637 ffi_call_unix64 + 79
3   _ctypes.cpython-35m-darwin.so 	0x000000010fe54e98 ffi_call + 872
4   _ctypes.cpython-35m-darwin.so 	0x000000010fe5004b _ctypes_callproc + 891
5   _ctypes.cpython-35m-darwin.so 	0x000000010fe4a301 PyCFuncPtr_call + 1057
6   libpython3.5m.dylib           	0x000000010e9d3570 PyObject_Call + 96
7   libpython3.5m.dylib           	0x000000010ea929a7 PyEval_EvalFrameEx + 25287
8   libpython3.5m.dylib           	0x000000010ea96815 _PyEval_EvalCodeWithName + 2421
9   libpython3.5m.dylib           	0x000000010ea9710f fast_function + 351
10  libpython3.5m.dylib           	0x000000010ea92829 PyEval_EvalFrameEx + 24905
11  libpython3.5m.dylib           	0x000000010ea9708f fast_function + 223
12  libpython3.5m.dylib           	0x000000010ea92829 PyEval_EvalFrameEx + 24905
13  libpython3.5m.dylib           	0x000000010ea9708f fast_function + 223
14  libpython3.5m.dylib           	0x000000010ea92829 PyEval_EvalFrameEx + 24905
15  libpython3.5m.dylib           	0x000000010ea96815 _PyEval_EvalCodeWithName + 2421
16  libpython3.5m.dylib           	0x000000010ea9710f fast_function + 351
17  libpython3.5m.dylib           	0x000000010ea92829 PyEval_EvalFrameEx + 24905
18  libpython3.5m.dylib           	0x000000010ea96815 _PyEval_EvalCodeWithName + 2421
19  libpython3.5m.dylib           	0x000000010ea9710f fast_function + 351
20  libpython3.5m.dylib           	0x000000010ea92829 PyEval_EvalFrameEx + 24905
21  libpython3.5m.dylib           	0x000000010ea96815 _PyEval_EvalCodeWithName + 2421
22  libpython3.5m.dylib           	0x000000010ea9710f fast_function + 351
23  libpython3.5m.dylib           	0x000000010ea92829 PyEval_EvalFrameEx + 24905
24  libpython3.5m.dylib           	0x000000010ea96815 _PyEval_EvalCodeWithName + 2421
25  libpython3.5m.dylib           	0x000000010ea9710f fast_function + 351
26  libpython3.5m.dylib           	0x000000010ea92829 PyEval_EvalFrameEx + 24905
27  libpython3.5m.dylib           	0x000000010ea96815 _PyEval_EvalCodeWithName + 2421
28  libpython3.5m.dylib           	0x000000010ea8c661 PyEval_EvalCode + 81
29  libpython3.5m.dylib           	0x000000010eabd911 PyRun_FileExFlags + 209
30  libpython3.5m.dylib           	0x000000010eabd0c3 PyRun_SimpleFileExFlags + 851
31  libpython3.5m.dylib           	0x000000010ead565a Py_Main + 3450
32  python                        	0x000000010e9c1e18 main + 216
[... 357 lines cut ...]
VM_ALLOCATE                        536K       22 
__DATA                            26.7M      286 
__FONT_DATA                          4K        2 
__GLSLBUILTINS                    5176K        2 
__LINKEDIT                       221.1M       45 
__TEXT                           183.0M      279 
__UNICODE                          564K        2 
mapped file                       55.4M       18 
shared memory                     2716K       10 
===========                     =======  ======= 
TOTAL                            686.7M      771 
TOTAL, minus reserved VM space   685.9M      771 

Model: MacBookPro12,1, BootROM 180.0.0.0.0, 2 processors, Intel Core i5, 2.7 GHz, 8 GB, SMC 2.28f7
Graphics: Intel Iris Graphics 6100, Intel Iris Graphics 6100, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1867 MHz, 0x80AD, 0x483943434E4E4E424C54414C41522D4E5544
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1867 MHz, 0x80AD, 0x483943434E4E4E424C54414C41522D4E5544
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x133), Broadcom BCM43xx 1.0 (7.77.61.1 AirPortDriverBrcmNIC-1305.2)
Bluetooth: Version 6.0.9f2, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0256G, 251 GB
USB Device: USB 3.0 Bus
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
```

alimuldal, 2019-02-25:
This works fine for me on Debian. @saran-t Did you have any luck reproducing this on a Mac?