google-deepmind/dm_control #75: Running viewer without an environment

tkelestemur, 2019-02-01:
I'm loading a custom MuJoCo environment and I want to run the viewer without defining a RL environment and task. Below is the way I load the XML file:

```
MODEL_PATH = PATH + '/models/data_collection.xml'

physics = mujoco.Physics.from_xml_path(MODEL_PATH)
pixels = physics.render()
```

Is this possible?

paksas, 2019-02-02:
Hello,

Currently the viewer supports only the environment visualisation.

Best regards,
Piotr Trochim

iandanforth, 2019-02-21:
@tkelestemur Can you describe your usecase? (If it is to view the model without additional requirements MuJoCo provides a viewer onto which you can drag an xml file.)

tkelestemur, 2019-02-21:
@iandanforth I can easily use MuJoCo's viewer but I wanna use dm_control as the python wrapper with a viewer that I can run from python code. I don't need to define an RL environment for my case.

iandanforth, 2019-02-22:
@tkelestemur How about this.

https://gist.github.com/iandanforth/04521ba10decdedc6c46579b9804a833

If you save both of those files to dm_control/dm_control/suite

You should then be able to use `xml-explore.py` like this:

`python xml-explore.py <yourfile>.xml`

Here's what it looks like running my muscled-ant.xml model with custom textures saved in the same dir.

[image]

If you compare `empty.py` with the original `cartpole.py` you should be able to add back in any functionality you need to get a real task running. Consider this a minimal starting point.

tkelestemur, 2019-02-25:
@iandanforth This worked great. The only drawback is that every time I change something in the `empty.py` I need to install dm_control again. Other than that this is all I needed.

Thanks a lot!

closed 2019-02-25 by tkelestemur