google-deepmind/dm_control #80: mocap_demo error (matplotlib problem?): [GLFWApplication _setup:]: unrecognized selector sent to instance 0x10f0b3ae0

bibbygoodwin, 2019-03-01:
Currently something in the package is resulting in this error when I run the mocap_demo script. The error seems to be the same as #[deepmind/dm_control#62](https://github.com/deepmind/dm_control/issues/62#issue-379954719), and given that the author of that issue put it down to something involving matplotlib it is possible that this is the root cause.

One of the replies to that issue from @saran-t (https://github.com/deepmind/dm_control/issues/62#issuecomment-438239756_) failed to reproduce the error because, it seems, they ran into the more standard matplotlib on OSX in virtualenvs problem, where a "ImportError: Python is not installed as a framework." error is thrown. If @saran-t sets the matplotlib backend to 'TkAgg' following the solution here: https://stackoverflow.com/questions/21784641/installation-issue-with-matplotlib-python then they should be able to reproduce this error?

Error Output:

visualizing cmu mocap data ...
2019-02-28 23:13:29.686 python[50314:14388105] -[GLFWApplication _setup:]: unrecognized selector sent to instance 0x10f0b3ae0
2019-02-28 23:13:29.693 python[50314:14388105] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[GLFWApplication _setup:]: unrecognized selector sent to instance 0x10f0b3ae0'
*** First throw call stack:
(
0 CoreFoundation 0x000000010010e2db __exceptionPreprocess + 171
1 libobjc.A.dylib 0x0000000100dbfc76 objc_exception_throw + 48
2 CoreFoundation 0x00000001001a6db4 -[NSObject(NSObject) doesNotRecognizeSelector:] + 132
3 CoreFoundation 0x0000000100084820 _**forwarding**_ + 1456
4 CoreFoundation 0x00000001000841e8 _CF_forwarding_prep_0 + 120
5 Tk 0x0000000110663318 TkpInit + 467
6 Tk 0x00000001105e2252 Tk_Init + 1710
7 _tkinter.cpython-35m-darwin.so 0x00000001104bfdcd Tcl_AppInit + 77
8 _tkinter.cpython-35m-darwin.so 0x00000001104bd7d9 _tkinter_create + 889
9 Python 0x00000001007b8f59 PyCFunction_Call + 233
10 Python 0x00000001008450e8 PyEval_EvalFrameEx + 36648
11 Python 0x0000000100845cf0 _PyEval_EvalCodeWithName + 2368
12 Python 0x0000000100845df7 PyEval_EvalCodeEx + 71
13 Python 0x0000000100793c7a function_call + 186
14 Python 0x0000000100761643 PyObject_Call + 99
15 Python 0x000000010077c64c method_call + 140
16 Python 0x0000000100761643 PyObject_Call + 99
17 Python 0x00000001007d6291 slot_tp_init + 81
18 Python 0x00000001007ccf24 type_call + 212
19 Python 0x0000000100761643 PyObject_Call + 99
20 Python 0x000000010083dfa6 PyEval_EvalFrameEx + 7654
21 Python 0x00000001008447a8 PyEval_EvalFrameEx + 34280
22 Python 0x0000000100845cf0 _PyEval_EvalCodeWithName + 2368
23 Python 0x0000000100845df7 PyEval_EvalCodeEx + 71
24 Python 0x0000000100793c7a function_call + 186
25 Python 0x0000000100761643 PyObject_Call + 99
26 Python 0x000000010084161e PyEval_EvalFrameEx + 21598
27 Python 0x0000000100845cf0 _PyEval_EvalCodeWithName + 2368
28 Python 0x0000000100844086 PyEval_EvalFrameEx + 32454
29 Python 0x00000001008447a8 PyEval_EvalFrameEx + 34280
30 Python 0x0000000100845cf0 _PyEval_EvalCodeWithName + 2368
31 Python 0x0000000100845e51 PyEval_EvalCode + 81
32 Python 0x000000010083967f builtin_exec + 623
33 Python 0x00000001007b8f59 PyCFunction_Call + 233
34 Python 0x00000001008450e8 PyEval_EvalFrameEx + 36648
35 Python 0x0000000100845cf0 _PyEval_EvalCodeWithName + 2368
36 Python 0x0000000100844086 PyEval_EvalFrameEx + 32454
37 Python 0x0000000100845cf0 _PyEval_EvalCodeWithName + 2368
38 Python 0x0000000100845df7 PyEval_EvalCodeEx + 71
39 Python 0x0000000100793c7a function_call + 186
40 Python 0x0000000100761643 PyObject_Call + 99
41 Python 0x000000010088c071 RunModule + 161
42 Python 0x000000010088cccb Py_Main + 2315
43 python 0x0000000100000e32 python + 3634
44 python 0x0000000100000c84 python + 3204
45 ??? 0x0000000000000007 0x0 + 7
)
libc++abi.dylib: terminating with uncaught exception of type NSException
./scripts/data_visualize.sh: line 7: 50314 Abort trap: 6 python -m data.vis_cmu_mocap --data_path ../data/cmu_mocap/all_asfamc --save_path ../data/cmu_mocap_vis

TianhongDai, 2019-03-27:
@bibbygoodwin Hi, what i suggest is trying to use `cv2` to do the visualisation
```python 
import cv2

for i in range(max_frame):
  cv2.imshow('demo', video[i])
  cv2.waitKey(100)
```

alimuldal, 2020-04-18:
Closing due to inactivity

closed 2020-04-18 by alimuldal