google-deepmind/dm_control #82: Documentation for composer

amandlek, 2019-03-06:
We've been trying out the composer interface. The API design looks fantastic, but we're running into some issues, and it has been difficult for us to try and resolve them without additional documentation for the composer. Is there any detailed documentation available for the composer?

saran-t, 2019-03-24:
Would you be able to tell us exactly which parts you are finding particularly problematic? This would help me target my documentation effort to the most useful place. (Sorry for the late reply!)

alimuldal, 2020-06-23:
I think a lot of the details are now covered in the [tutorial colab](https://colab.research.google.com/github/deepmind/dm_control/blob/master/tutorial.ipynb) (31a3600870db8c8190c61eda1bd358f402e62d54)

redwrasse, 2020-11-30:
A formal readthedocs would be nice, if this library is still active.