google-deepmind/dm_control #86: How to control the randomness ?

zuoxingdong, 2019-04-09:
It seems to me the environment is deterministic, e.g.

```python
env = suite.load(domain_name='cheetah', task_name='run')
obs1 = env.reset().observation
obs2 = env.reset().observation
print(obs1)
print(obs2)
```
gives 
```
OrderedDict([('position', array([-0.10657491,  0.03791366, -0.01320079,  0.02708236, -0.05382936,
        0.00701578, -0.07508983, -0.06376804])), ('velocity', array([-0.00637606, -0.00594656,  0.00852484, -0.00541404,  0.00263544,
       -0.01481148, -0.00058044, -0.03207024, -0.03622259]))])
OrderedDict([('position', array([-0.10657491,  0.03791366, -0.01320079,  0.02708236, -0.05382936,
        0.00701578, -0.07508983, -0.06376804])), ('velocity', array([-0.00637606, -0.00594656,  0.00852484, -0.00541404,  0.00263544,
       -0.01481148, -0.00058044, -0.03207024, -0.03622259]))])
```
- Shouldn't we expect each call to `reset` randomly initializes the state ?
- Tuning `env.task.random.seed(99)` does not change the result.

alimuldal, 2019-04-10:
Well spotted. The initial state of the `cheetah` environment is actually deterministic - you can see there's no randomization going on in [`initialize_episode`] at the moment. This should probably be considered a bug!

All of the other environments in the control suite have randomized initial states. You can control the seed by passing `task_kwargs={'random': seed_or_randomstate}` to `suite.load`, for example:
```python
from dm_control import suite

seed = 42
env = suite.load(domain_name='cartpole', task_name="balance", task_kwargs={'random': seed})
print(env.reset().observation)
# OrderedDict([('position', array([-0.02509198,  0.99953037,  0.03064377])),
#              ('velocity', array([ 0.00647689,  0.0152303 ]))])
print(env.reset().observation)
# OrderedDict([('position', array([-0.06879627,  0.99972641, -0.02339024])),
#              ('velocity', array([ 0.01579213,  0.00767435]))])

# Reload with the same seed
env = suite.load(domain_name='cartpole', task_name="balance", task_kwargs={'random': seed})
print(env.reset().observation)  # Same as first
# OrderedDict([('position', array([-0.02509198,  0.99953037,  0.03064377])),
#              ('velocity', array([ 0.00647689,  0.0152303 ]))])
```
[`initialize_episode`]: https://github.com/deepmind/dm_control/blob/master/dm_control/suite/cheetah.py#L78-L86

zuoxingdong, 2019-04-11:
@alimuldal Thanks a lot for this information ! Indeed, for other environments, the initial state is randomly sampled (controllable via seed arg)

Will it make sense to fix cheetah-run to also sample initial state ? In `cheetah.py` there is one comment `# Stabilize the model before the actual simulation` which runs 200 steps in the internal engine to stabilize the cheetah, I'm wondering a small perturbation of initial state can still make cheetah survive well in this first 200 physical steps, as long as the perturbation is not significant.

closed 2019-04-15 by alimuldal

davidireland3, 2023-04-18:
> Well spotted. The initial state of the `cheetah` environment is actually deterministic - you can see there's no randomization going on in [...]

Is this still the case that to reset to a certain seed we have to re-load the environment again? is there no option to set a seed in e.g. the `reset` call?

jacr13, 2024-04-15:
There is a way to reset to a given seeding without re-creating the environment:
```
from dm_control import suite
import numpy as np

seed = 42
env = suite.load(domain_name='cartpole', task_name="balance", task_kwargs={'random': seed})
print(env.reset().observation)
# OrderedDict([('position', array([-0.02509198,  0.99953037,  0.03064377])),
#              ('velocity', array([ 0.00647689,  0.0152303 ]))])

print(env.reset().observation)
# OrderedDict([('position', array([-0.06879627,  0.99972641, -0.02339024])),
#              ('velocity', array([ 0.01579213,  0.00767435]))])

# Change task random state with the desired seed
env.task._random = np.random.RandomState(seed)
print(env.reset().observation)  # Same as first
# OrderedDict([('position', array([-0.02509198,  0.99953037,  0.03064377])),
#              ('velocity', array([ 0.00647689,  0.0152303 ]))])
```

commit 19e998b5 (2019-04-15): Randomize the cheetah's internal joints at the start of each episode

This randomization has quite a subtle effect on the initial state since we run the simulation forward for 2 seconds at the start of each episode, allowing the joints to settle in a similar position across episodes.

I also added a regression test that checks that two consecutive initial states have different observations.

Closes #86

PiperOrigin-RevId: 243584776