google-deepmind/dm_control #87: Cannot launch the viewer using ssh on macos Mojave

ArchGentoo, 2019-04-11:
I tried to run the following code on a remote ubuntu 18.04 server using ssh and the local machine's OS is MacOS Mojave. Btw. xclock and xeyes running correctly using ssh. The server setup is Python 3.5.6 with anaconda env. Also, the following codes can run correctly on the server. I don't know where the problem is and try to search answers for 'How to do X11 Forwarding on a Mac' but can't help. What to do next?
`from dm_control import viewer
  from dm_control import suite
  env = suite.load(domain_name="humanoid", task_name="stand")
  viewer.launch(env)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/cong/anaconda3/envs/dm_control/lib/python3.5/site-packages/dm_control/viewer/__init__.py", line 42, in launch
    app = application.Application(title=title, width=width, height=height)
  File "/home/cong/anaconda3/envs/dm_control/lib/python3.5/site-packages/dm_control/viewer/application.py", line 191, in __init__
    self._window = gui.RenderWindow(width, height, title)
  File "/home/cong/anaconda3/envs/dm_control/lib/python3.5/site-packages/dm_control/viewer/gui/__init__.py", line 35, in ErrorRenderWindow
    'Cannot create a window because no windowing system could be imported')
ImportError: Cannot create a window because no windowing system could be imported`

alimuldal, 2019-04-11:
The viewer is not going to work over SSH because it uses the GLFW rendering backend, which requires GLX. `ssh -X` invokes indirect rendering, which generally only supports OpenGL versions up to 1.4. MuJoCo itself needs at least OpenGL 1.5 with the `ARB_framebuffer_object` extension [see here][mujoco opengl].

If you must run the viewer remotely, your best option might be to run it in a remote desktop session (e.g. VNC).

[mujoco opengl]: http://www.mujoco.org/book/programming.html#inOpenGL

closed 2019-04-11 by alimuldal

ArchGentoo, 2019-04-11:
Thanks a lot!