google-deepmind/dm_control #90: Support pip installation

zuoxingdong, 2019-04-14:
It'll be very convenient to support pip installation via PyPI

alimuldal, 2019-05-10:
`dm_control` can be installed using `pip` by passing the address to our Github repository:
```bash
pip install git+git://github.com/deepmind/dm_control.git
```
Alternatively you could check out a local copy of `dm_control`, and then install it by path
```bash
pip install /path/to/dm_control
```
I suspect what you're really asking is whether we can host `dm_control` on PyPI  (so you could just do `pip install dm_control`)? We're looking into that, but there are some complications due to the fact that we auto-generate the ctypes bindings from the MuJoCo headers as part of the installation process.

saran-t, 2019-12-24:
`dm_control` is now available on PyPI.

closed 2019-12-24 by saran-t