google-deepmind/dm_control #92: I got such error 'can't pickle weakref objects' when I run  multiprocessing program.

Gaoee, 2019-04-23:
I want run the soccer environment in multiprocessing program. But I got such error 'can't pickle weakref objects'. I know it is beacause the 'env' I created is weakref, so I got such error when I run multiprocessing program. Are there any methods can be used to tackle it?
Thank you!

alimuldal, 2019-04-23:
Could you be more specific about how you're trying to use multiprocessing? Composer environments aren't picklable, so you can't pass in the environment as an argument to the function you're running in each process.

One thing that should work is to instantiate a separate environment within each child process, e.g.
```python
from dm_control.locomotion import soccer
import multiprocessing

def create_and_step_environment(unused_seed):
  timesteps = []
  env = soccer.load(4)
  action = [player_spec.generate_value() for player_spec in env.action_spec()]
  for _ in range(20):
    timesteps.append(env.step(action))
  return timesteps

thread_pool = multiprocessing.Pool(5)
results = thread_pool.map(create_and_step_environment, range(10))
```

Gaoee, 2019-04-24:
Thank you for your advice. But I got another when I run your proposed code. The error is 'fatal IO error 11 (Resource temporarily unavailable)' which should change multiprocessing's start method to spawn:  `mp.set_start_method('spawn')` to solve.
[https://github.com/openai/mujoco-py/issues/40](url)

alimuldal, 2019-04-24:
I think that error is probably specific to GLFW which requires X11, whereas OSMesa and EGL don't. I suspect the code above would work without changing the spawn method if you set the environment variable `MUJOCO_GL=osmesa` when launching your interpreter.

Gaoee, 2019-04-25:
Yes, you are right. I have tested it and it works. Thank you.

closed 2019-04-25 by Gaoee