google-deepmind/dm_control #93: Access to reward function

theophilee, 2019-05-01:
Hi, first of all thanks for a great set of benchmarks!

How could I access the reward function? I'm working on some model-based RL project and I need to compute the reward function of arbitrary transitions. In open ai gym environments it is quite straightforward to figure out the reward, but here it seems more intricate.

Thanks a lot!

wpumacay, 2019-05-02:
Hi @theophilee. Regarding your question, you want to directly access the function and do a call like r(s,a,s') to obtain the reward?. Or do you just want to know how rewards are computed, and in which part of the code is the reward being computed?.

For the later, you can check each environment from the suite and check the extended Task class for that environment, e.g. [here](https://github.com/deepmind/dm_control/blob/977d47e042713f86bae179fb017a7849446f59bd/dm_control/suite/walker.py#L143) for the walker environment.

For the former, you could access the [*_task*](https://github.com/deepmind/dm_control/blob/977d47e042713f86bae179fb017a7849446f59bd/dm_control/rl/control.py#L63) variable from the environment, and call the get_reward function. Unfortunately, I don't think you could directly call it with the observations you get from the step function, as those are observations, and not the full internal state of the environment, which is the one used to compute the reward in most cases. You could still kind of force it by passing an object with the exact requirements to the reward function, kind of like faking the internal state.

theophilee, 2019-05-02:
Thanks a lot for your quick reply Wilbert, I'd like to evaluate r(s,a,s') ideally in parallel for many transitions (to evaluate a bunch of plans generated by a particle-based planner). Is that possible? If it's not it makes it quite hard to do any model-based planning with the environment suite...

One way to compute r(s,a,s') would be to reset the state to s with 
        with env.physics.reset_context():
            env.physics.data.qpos[:] = state[0]
            env.physics.data.qvel[:] = state[1]
and use step() to get the reward but this is not parallel and super inefficient.
Do you see any better way?

wpumacay, 2019-05-02:
Hi @theophilee . I'm not 100% sure, but I think that is not quite possible. Perhaps the authors could help you a bit more with this one. 

I haven't got much experience with model-based RL, but still there are some subtleties with your requirements that I think don't apply for these benchmarks.

* If I understand correctly (based also in your post in the [gym](https://github.com/openai/gym/issues/1463) repo) you would also want the environment model for the application your are doing. If that's the case, then I don't think benchmarks for RL will suffice, because you don't have access to the model of the world, but can only query the simulator step by step to get experiences. Perhaps another framework (like [drake](https://github.com/RobotLocomotion/drake)) would be better suited for the job.

* Also (if I'm not mistaken), with model-based **RL** usually you want to first try to fit a model of the world from experiences, and then you query that model (you run your planner in that learned model). There you could run as many queries as particles you have, but here I think you are limited by the queries you have to make to the simulator to get the transitions, which in turn give you the rewards.

* Finally, the rewards are computed from specific internals of the state of the environment, and in very different ways (depending of each kind of environment). If you set the physics datastructure to a certain state, I think you then are forced to take a step in the simulator for the physics state be propagated to all the other internals used by the reward function calculation, like the torso z position or the horizontal velocity (in the walker env.). For some you could just take a step of forward kinematics (like xyz pos of the torso), but that would not suffice for all components of the reward function. So, I think your bottleneck is still the required step in the simulator.

theophilee, 2019-05-02:
Ideally I would have access to both the dynamics model T(s'|s,a) and the reward function r(s,a,s') in order to plan with the true dynamics/reward (as a baseline). If we learn an approximate dynamics model, we only need the reward function (what I want to do). 

I don't think either gym or dm-control provide easy access to the dynamics model, but gym does provide access to the reward function. I guess here the reward computation is more complex (based on internal states).

Thank you for your help @wpumacay. Any input  dm-control team, @alimuldal?

alimuldal, 2019-05-02:
As Wilbert says, the definition of the "state" input to the reward function really depends on the specifics of the environment. We always pass in the whole `Physics` instance to `get_reward`, so in general the reward can be a function of any of the fields within `Physics` (or even potentially state that lives outside of `Physics` such as attributes of the `Task`, although this isn't a pattern that we use or recommend). This design allows for maximum flexibility in terms of how reward functions can be defined.

The dynamics model is really just the MuJoCo simulation. We provide access to this via our Python bindings for MuJoCo (`env.physics`). If you're interested in the specifics of how MuJoCo works then the [Computation chapter](http://www.mujoco.org/book/computation.html) in the MuJoCo docs has a pretty comprehensive explanation. In order to compute state transitions you will, in general, need to step the simulation forward.

closed 2019-05-02 by alimuldal