google-deepmind/dm_control #95: How to use on Ubuntu 18? [Installation fails despite same steps as my Ubuntu 16 system]

DanielTakeshi, 2019-05-10:
I have two different machines, one with Ubuntu 16.04 (and Python 3.5.2) and another with Ubuntu 18.04 (and Python 3.6.7). I can get `dm_control` working on the 16.04 machine but not the 18.04 machine. The README here implies that Ubuntu 16 is the most tested version, so I'm not sure if 18 is supposed to work, but is it _possible_ to get it working? Here are my steps to reproduce, and hopefully the error message will be clear.

# Ubuntu 16.04 machine

MuJoCo 2.0 is installed here with mjkey at the standard location.

```
daniel@takeshi:~ $ ls -lh .mujoco/
total 8.0K
-rw-rw-r-- 1 daniel daniel  965 Mar  8 13:18 mjkey.txt
drwxrwxr-x 7 daniel daniel 4.0K Mar  8 12:17 mujoco200_linux
```

I do `pip install .` (NOT in editable mode) **in a clean Python 3.5.2 virtualenv**, as of master commit 37a17bf3d6800faee8432fd3ce9df8f4c1725e7f and it seems to be working. (It was already installed before this command, I'm just doing it again to confirm.)

```
(mujoco-py3) daniel@takeshi:~/dm_control (master) $ pip install .
Processing /home/daniel/dm_control
Requirement already satisfied: absl-py>=0.7.0 in /home/daniel/Envs/mujoco-py3/lib/python3.5/site-packages (from dm-control==0.0.0) (0.7.1)
Requirement already satisfied: enum34 in /home/daniel/Envs/mujoco-py3/lib/python3.5/site-packages (from dm-control==0.0.0) (1.1.6)
Requirement already satisfied: future in /home/daniel/Envs/mujoco-py3/lib/python3.5/site-packages (from dm-control==0.0.0) (0.17.1)
Requirement already satisfied: futures in /home/daniel/Envs/mujoco-py3/lib/python3.5/site-packages (from dm-control==0.0.0) (3.1.1)
Requirement already satisfied: glfw in /home/daniel/Envs/mujoco-py3/lib/python3.5/site-packages (from dm-control==0.0.0) (1.7.1)
Requirement already satisfied: lxml in /home/daniel/Envs/mujoco-py3/lib/python3.5/site-packages (from dm-control==0.0.0) (4.3.3)
Requirement already satisfied: numpy in /home/daniel/Envs/mujoco-py3/lib/python3.5/site-packages (from dm-control==0.0.0) (1.16.3)
Requirement already satisfied: pyopengl in /home/daniel/Envs/mujoco-py3/lib/python3.5/site-packages (from dm-control==0.0.0) (3.1.0)
Requirement already satisfied: pyparsing in /home/daniel/Envs/mujoco-py3/lib/python3.5/site-packages (from dm-control==0.0.0) (2.4.0)
Requirement already satisfied: setuptools in /home/daniel/Envs/mujoco-py3/lib/python3.5/site-packages (from dm-control==0.0.0) (41.0.1)
Requirement already satisfied: six in /home/daniel/Envs/mujoco-py3/lib/python3.5/site-packages (from dm-control==0.0.0) (1.12.0)
Building wheels for collected packages: dm-control
  Building wheel for dm-control (setup.py) ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-kkd65c2o/wheels/29/a9/b5/bb7d9be35d3d72476e5db2fc741bdc8e8d4ee9f4ca6a96f9f5
Successfully built dm-control
Installing collected packages: dm-control
  Found existing installation: dm-control 0.0.0
    Uninstalling dm-control-0.0.0:
      Successfully uninstalled dm-control-0.0.0
Successfully installed dm-control-0.0.0
```

I decide to run a python script. I go into `dm_control/viewer` and run this command:

```
(mujoco-py3) daniel@takeshi:~/dm_control/dm_control/viewer (master) $ python viewer_test.py
```

and it seems to work!

I don't set any environment variables to do anything MuJoCo related because I installed them according to the defaults of what this repository expects. For example, my LD_LIBRARY_PATH says nothing about MuJoCo, etc.

# Ubuntu 18.04 machine

MuJoCo 2.0 is installed here with the mjkey at the standard location:

```
seita@hermes1-ubuntu18:~ $ ls -lh .mujoco/
total 8.0K
-rw-r--r-- 1 seita seita  965 May 10 14:41 mjkey.txt
drwxrwxr-x 7 seita seita 4.0K Oct  1  2018 mujoco200_linux
```

And furthermore, here's the library files:

```
seita@hermes1-ubuntu18:~ $ ls -lh .mujoco/mujoco200_linux/bin/
total 6.0M
-rwxrwxr-x 1 seita seita  19K Oct  1  2018 basic
-rwxrwxr-x 1 seita seita  14K Oct  1  2018 compile
-rwxrwxr-x 1 seita seita  19K Oct  1  2018 derivative
-rwxrwxr-x 1 seita seita 802K Oct  1  2018 libglewegl.so
[... 82 lines cut ...]
/home/seita/.mujoco/
(mujoco-py3) seita@hermes1-ubuntu18:~/dm_control/dm_control/viewer (master) $ echo $MJLIB_PATH 
/home/seita/.mujoco/mujoco200_linux/bin
(mujoco-py3) seita@hermes1-ubuntu18:~/dm_control/dm_control/viewer (master) $ python viewer_test.py 
WARNING: Logging before flag parsing goes to stderr.
W0510 15:54:07.450641 140009134028608 __init__.py:36] mjbindings failed to import mjlib and other functions. libmujoco.so may not be accessible.
Traceback (most recent call last):
  File "viewer_test.py", line 22, in <module>
    from dm_control.mujoco.wrapper.mjbindings import enums
  File "/home/seita/Envs/mujoco-py3/lib/python3.6/site-packages/dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/home/seita/Envs/mujoco-py3/lib/python3.6/site-packages/dm_control/mujoco/engine.py", line 44, in <module>
    from dm_control.mujoco import index
  File "/home/seita/Envs/mujoco-py3/lib/python3.6/site-packages/dm_control/mujoco/index.py", line 92, in <module>
    from dm_control.mujoco.wrapper import util
  File "/home/seita/Envs/mujoco-py3/lib/python3.6/site-packages/dm_control/mujoco/wrapper/__init__.py", line 25, in <module>
    from dm_control.mujoco.wrapper.core import callback_context
  File "/home/seita/Envs/mujoco-py3/lib/python3.6/site-packages/dm_control/mujoco/wrapper/core.py", line 32, in <module>
    from dm_control.mujoco.wrapper.mjbindings import functions
  File "/home/seita/Envs/mujoco-py3/lib/python3.6/site-packages/dm_control/mujoco/wrapper/mjbindings/functions.py", line 21, in <module>
    mjlib = util.get_mjlib()
  File "/home/seita/Envs/mujoco-py3/lib/python3.6/site-packages/dm_control/mujoco/wrapper/util.py", line 124, in get_mjlib
    raise OSError("Cannot find MuJoCo library at {}.".format(library_path))
OSError: Cannot find MuJoCo library at /home/seita/.mujoco/mujoco200_linux/bin.
```

alimuldal, 2019-05-11:
If you're setting `MJLIB_PATH`, this needs to be the path to the `.so` library rather than the directory, e.g. `MJLIB_PATH=/home/seita/.mujoco/mujoco200_linux/bin/libmujoco200.so`. Same for `MJKEY_PATH`.

DanielTakeshi, 2019-11-14:
Sorry for the late response @alimuldal ! Making the modification to those two env variables allows the viewer script I ran in my last post to work. Not sure if this means the repo is installed correctly but at least the test is passing.

```
$ echo $MJKEY_PATH 
/home/seita/.mujoco/mjkey.txt
$ echo $MJLIB_PATH 
/home/seita/.mujoco/mujoco200_linux/bin/libmujoco200.so
```

closed 2019-11-14 by DanielTakeshi