google-deepmind/dm_control #96: Parallelized training generates many 'C+G' type processes on GPU 0

zuoxingdong, 2019-05-13:
I'm parallelizing my implementation of RL algorithms with dm_control (master branch), it seems many `C+G` type processes are generated and put on GPU0, according to what `nvidia-smi` shows. But when I replace dm_control enviroment with OpenAI gym Mujoco environment, it does not have this behavior anymore. So probably the problem is not related to DL framework itself.

Is there a potential reason why it does this ?

alimuldal, 2019-05-13:
The only thing `dm_control` could possibly be using the GPU for is rendering. Which [rendering backend](https://github.com/deepmind/dm_control#rendering) are you using? Do you get the same behaviour if you force software rendering (`MUJOCO_GL=osmesa`)?

zuoxingdong, 2019-05-23:
Hi @alimuldal , thanks a lot for the information. Yes, it disappears after setting `MUJOCO_GL=osmesa`. However, the remaining question is I didn't have any call to `render`, but just `step`, why it still occupies GPU with `C+G` process when rendering is by default ?

alimuldal, 2019-05-23:
We still create an OpenGL context each time `Physics` is instantiated - that probably explains why you're seeing those processes on the GPU. These will probably waste a little bit of GPU memory if you're not doing any rendering, but other than that I wouldn't expect them to use much in the way of resources. For now the easiest way to avoid this is to use OSMesa.

zuoxingdong, 2019-05-24:
Thanks @alimuldal , I'll stick with OSMesa for now. Just for further situation, in my case, I'm parallelizing experiment with dozens of processes, the GPU memory occupied for rendering could rise up to a few GBs.

closed 2019-06-10 by alimuldal