google-deepmind/dm_control #97: Error importing mjlib when importing suite

matthioe, 2019-05-28:
Hi,
I'm currently having trouble importing the suite in python. When I do "from dm_control import suite", I get the following error:
`>>> from dm_control import suite
WARNING: Logging before flag parsing goes to stderr.
W0528 10:33:53.116941 140242758797120 __init__.py:36] mjbindings failed to import mjlib and other functions. libmujoco.so may not be accessible.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/work/scratch/ey25ocyd/newenv/lib/python3.6/site-packages/dm_control/suite/__init__.py", line 28, in <module>
    from dm_control.suite import acrobot
  File "/work/scratch/ey25ocyd/newenv/lib/python3.6/site-packages/dm_control/suite/acrobot.py", line 24, in <module>
    from dm_control import mujoco
  File "/work/scratch/ey25ocyd/newenv/lib/python3.6/site-packages/dm_control/cp/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/work/scratch/ey25ocyd/newenv/lib/python3.6/site-packages/dm_control/mujoco/engine.py", line 44, in <module>
    from dm_control.mujoco import index
  File "/work/scratch/ey25ocyd/newenv/lib/python3.6/site-packages/dm_control/mujoco/index.py", line 92, in <module>
    from dm_control.mujoco.wrapper import util
  File "/work/scratch/ey25ocyd/newenv/lib/python3.6/site-packages/dm_control/mujoco/wrapper/__init__.py", line 25, in <module>
    from dm_control.mujoco.wrapper.core import callback_context
  File "/work/scratch/ey25ocyd/newenv/lib/python3.6/site-packages/dm_control/mujoco/wrapper/core.py", line 33, in <module>
    from dm_control.mujoco.wrapper.mjbindings import mjlib
ImportError: cannot import name 'mjlib'
>>>  mjbindings failed to import mjlib and other functions. libmujoco.so may not be accessible.`

Installation via pip worked without errors, I've also set the LD_LIBRARY_PATH. I even tried setting the MJLIB_PATH to libmujoco.so myself but I still get the error. It seems like it finds the file but has problems importing it.

The problem could be related to [this issue](https://github.com/deepmind/dm_control/issues/3) as I haven't installed any rendering backend (GLFW/GLEW/OSMesa/...). As I am using it on a cluster without windowing system and won't use any rendering functionality, this should be no problem, right? I also followed [this hint](https://github.com/deepmind/dm_control/issues/20#issuecomment-360988234), executing
`os.environ['DISABLE_MUJOCO_RENDERING'] = '1'`
before the import but that also doesn't help.

matthioe, 2019-05-29:
A solution for [different problem](https://github.com/deepmind/dm_control/issues/76#issuecomment-463485054) finally did it for me.
I just run `export MUJOCO_GL="osmesa"` before doing the imports and now everything seems to work.

Would be nice to mention this setting in the readme or so for people who don't want to use rendering / don't have any rendering backend installed.

closed 2019-05-29 by matthioe

episodeyang, 2020-01-04:
For those who are using a headless linux box without X-display,  use this:

```bash
      export MUJOCO_GL=osmesa
      export MJLIB_PATH=$HOME/.mujoco/mujoco200/bin/libmujoco200.so
      export MJKEY_PATH=$HOME/.mujoco/mujoco200/mjkey.txt
      export LD_LIBRARY_PATH=$HOME/.mujoco/mujoco200/bin:$LD_LIBRARY_PATH
      export MUJOCO_PY_MJPRO_PATH=$HOME/.mujoco/mujoco200/
      export MUJOCO_PY_MJKEY_PATH=$HOME/.mujoco/mujoco200/mjkey.txt
```