google-deepmind/dm_control #101: Why aren't mujoco paths consistent with mujoco-py?

ilinojkovic, 2019-06-11:
As said in the title, why are there differences between the two? It makes installation several steps more complicated, especially in the case when both need to be installed as dependencies to a third package. This is what I'm facing now when trying to install https://github.com/rlworkgroup/garage.

alimuldal, 2019-06-11:
You're referring to the fact that we look for MuJoCo in  `~/.mujoco/mujoco200_{macos,linux,win64}`, whereas `mujoco-py` looks in `~/.mujoco/mujoco200`? We're just using the same top-level directory names as in the vendor's zipfiles, e.g. you can install MuJoCo just by unzipping into `~/.mujoco`:
```bash
wget -q https://www.roboti.us/download/mujoco200_linux.zip -O /tmp/mujoco.zip && \
unzip -q /tmp/mujoco.zip -d ~/.mujoco/ && rm /tmp/mujoco.zip
```
For whatever reason `mujoco-py` decided to do their own thing with the directory names. If you prefer their naming convention then you can set `MJLIB_PATH` to point somewhere else. There's also a `--headers-dir` flag you can pass to `setup.py install` in order to override the default location of the MuJoCo headers that are used to auto-generate the ctypes bindings. I'd be open to the idea of making this an environment variable rather than a flag if that would be more convenient, but I'd prefer to stick to the vendor's naming scheme for the MuJoCo subdirectory.

ilinojkovic, 2019-06-11:
Hi @alimuldal, thanks for the reply!

Actually, I wasn't referring to the default paths, but the environment variables in case of a custom path. You use `MJKEY_PATH`, while they use `MUJOCO_PY_MJKEY_PATH`, and `MJLIB_PATH` and `--headers-dir` can be `$MUJOCO_PY_MUJOCO_PATH/bin` and `$MUJOCO_PY_MUJOCO_PATH/include` respectively.

I'm aware that `mujoco-py` is not an official project, and therefore you don't need to be compliant with their naming, but it seems to me this package is by far the most used python wrapper for mujoco, so making a smoother integration imho helps both projects.

alimuldal, 2019-06-12:
Ah, you just mean that the names of the environment variable are different?

I see that as a feature rather than a bug - `mujoco-py`'s environment variables shouldn't interfere with the behaviour of `dm_control` or vice versa. Presumably that's why `mujoco-py` prefixed their variable names with the name of their library to avoid potential clashes. With hindsight it might have been better for us to do the same and use `DM_CONTROL_MJKEY_PATH` instead of just `MJKEY_PATH` etc., but these names predate the existence of the `dm_control` package, and changing them now would inconvenience our current users. 

For your use-case it's fairly trivial to set these variables to the same value if that's the behaviour you're looking for, e.g. `export {MUJOCO_PY_}MJKEY_PATH=~/.mujoco/some/custom/key.txt`.

closed 2019-06-12 by alimuldal

ilinojkovic, 2019-06-13:
I see. Thanks for the response.