google-deepmind/dm_control #104: ubuntu 18 dm_control.viewer error

adol001, 2019-06-16:
I have installed dm_control in ubuntu 18.04, but I can't use viewer, other parts seem ok
`  File "/home/adol/anaconda3/lib/python3.6/site-packages/dm_control/viewer/gui/__init__.py", line 35, in ErrorRenderWindow
    'Cannot create a window because no windowing system could be imported')
ImportError: Cannot create a window because no windowing system could be imported`

I have installed current nvidia driver, and other dependency (sudo apt install libglfw3 libglew2.0 libgl1-mesa-glx libosmesa6)

adol001, 2019-06-16:
I have a successful installation on ubuntu 16.04, on ubuntu 18.04, there is a bug in glew or linux, you can fix it by appending  "fs.inotify.max_user_watches=32768" to /etc/sysctl.d/99-sysctl.conf, the default is smaller than ten thousands

closed 2019-06-16 by adol001

DanielTakeshi, 2019-11-22:
@adol001  Thanks, is it OK to show your `/etc/sysctl.d/99-sysctl.conf` file? Did you just add that line at the  bottom?

adol002, 2019-11-25:
> @adol001 Thanks, is it OK to show your `/etc/sysctl.d/99-sysctl.conf` file? Did you just add that line at the bottom?

I am adol001, yes

adol002, 2019-11-25:
> @adol001 Thanks, is it OK to show your `/etc/sysctl.d/99-sysctl.conf` file? Did you just add that line at the bottom?

I am adol001, yes, just add that line.