google-deepmind/dm_control #106: Changing physics.model attributes?

MishaLaskin, 2019-06-28:
I'm wondering if there's an API to change model attributes. For example, take the point mass env - I'd like to change the point mass's color depending on some other variable (e.g. it's position, proximity to target, or reward). Is it possible to do this?

When I try to set `physics.model.{some_attribute} = {some_value}` explicitly I get a `AttributeError: can't set attribute` error.

alimuldal, 2019-07-01:
Hi Misha,

Most of the properties of `physics.model` and `physics.data` are numpy arrays that expose direct views onto the underlying C structs used by MuJoCo. To mutate them you should write into the contents of the array rather than assigning over the top of it. For example:
```python
from dm_control import suite

env = suite.load('cartpole', 'swingup')
print(repr(env.physics.data.qvel))  # (nq,) array of joint velocities
# array([ 0.,  0.])
env.physics.data.qvel[:] = 0.1, 0.2  # Set the velocities of the slide and hinge joints
# env.physics.data.qvel = 0.1, 0.2  # This won't work (AttributeError)
```
To set just the velocity of the first joint you could write into `qvel[0]` and so forth. Addressing elements by their indices can be error-prone, so we also provide a way to address them by their names given in the XML. For example:
```python
env.physics.named.data.qvel['slider'] = -0.5
print(repr(env.physics.named.data.qvel))
# FieldIndexer(qvel):
# 0  slider [-0.5     ]
# 1 hinge_1 [ 0.2     ]
```
See section 5 in our [tech report] and the task implementations in the [`dm_control/suite`] directory for more examples. I'd also recommend taking a look at the [MuJoCo docs] in case you're unfamiliar with MuJoCo's API (and in particular the distinction between `model` and `data`).

[tech report]: https://arxiv.org/pdf/1801.00690.pdf
[`dm_control/suite`]: https://github.com/deepmind/dm_control/tree/master/dm_control/suite
[MuJoCo docs]: http://mujoco.org/book/index.html

closed 2019-07-01 by alimuldal

MishaLaskin, 2019-07-01:
Got it thank you!