google-deepmind/dm_control #108: Mixed scalar and vector observations

hartikainen, 2019-08-05:
Some of the environments (e.g. `domain_name='quadruped', task_name='run'`) have mixed scalar and vector observations. For example, `env.observation_spec()['torso_upright'].shape == ()` and `env.observation_spec()['torso_velocity'].shape == (3, )`. This is a bit confusing since certain operations like `np.concatenate` require all the inputs to be of the same shape and thus some manual handling needs to happen for scalar fields to reshape then into shape `(1, )`.

Is having scalar observations with shape `()` intentional? Would you accept a PR that converts all the scalar observations to be of shape `(1, )` within the environments?

alimuldal, 2019-08-06:
Hi Kristian,

Firstly, were you aware of the [`flat_observation`](https://github.com/deepmind/dm_control/blob/master/dm_control/rl/control.py#L44) kwarg to `control.Environment`? If you set this to `True` (e.g. `suite.load('cartpole', 'balance', environment_kwargs={'flat_observation': True}`) then the environment will return observations that have been flattened and concatenated into a single 1D vector. I suspect that might be what you're looking for.

Secondly, if you want to do your own thing in terms of concatenation you can use `np.hstack` in place of `np.concatenate`, which will deal gracefully with mixed sequences of 1D arrays  and scalars.

I'm a bit reluctant to change the default shapes of the observation arrays since we have quite a lot of existing code that depends on the current observation shapes.

hartikainen, 2019-08-07:
Hey @alimuldal,

Thanks for the quick reply. I am actually aware of the flat observation, but that doesn't fully work in my case since I preferable store the observations on the algorithms side as dictionaries. After thinking of this again, maybe this is more of a defect on our side rather than in `dm_control`.

> Secondly, if you want to do your own thing in terms of concatenation you can use np.hstack in place of np.concatenate, which will deal gracefully with mixed sequences of 1D arrays and scalars.

Neat, I did not know this. Thanks for the tip!

I'll close this for now.

closed 2019-08-07 by hartikainen