google-deepmind/dm_control #109: Action scale wrapper

hartikainen, 2019-08-05:
It would be nice if there was a wrapper that would allow actions to be scaled within a certain interval, similar to what [softlearning](https://github.com/rail-berkeley/softlearning/blob/master/softlearning/environments/gym/wrappers/normalize_action.py) and [gym](https://github.com/openai/gym/pull/1491) has for gym envs.

alimuldal, 2019-08-06:
With the exception of `quadruped` and `lqr`, all of the environments in the control suite accept actions scaled between -1 and 1 (`env.action_spec()` returns a `specs.BoundedArray` object describing the shape, dtype and bounds of the action array expected by the environment). A wrapper for rescaling actions might be a nice thing to have, though.

hartikainen, 2019-08-07:
Yeah, the `quadruped` env is the main issue for us. I implemented a wrapper in the [softlearning repo](https://github.com/hartikainen/softlearning/commit/6f71b5cfb7d4733a2e0f120dcef1f1e13b6cdf93) and can submit a PR for it sometime soon.

closed 2019-12-07 by hartikainen