google-deepmind/dm_control #114: Pre-allocated contact buffer is full, physics state is invalid.

RaveChase, 2019-08-27:
I encountered several warnings each time I run the dm_control.locomotion example as below:
W0827 .544679 139749952841536 core.py:299] Pre-allocated contact buffer is full. Increase nconmax above 100. Time = 0.0000.
W0827 .545073 139749952841536 core.py:85] Pre-allocated contact buffer is full. Increase nconmax above 100. Time = 0.0000.
W0827 .546310 139749952841536 engine.py:323] Physics state is invalid. Warning(s) raised: mjWARN_CONTACTFULL
And then the episode will terminate with time lasts 0.5s only. Such time is not expected obviously. There were similar problems about this on github. I've read some of them and followed their instructions. However, I failed to follow the variable to find where it was settled or changed, both in code or all .xml files.
So, where can I increase that? Or anyway else can solve this problem?

saran-t, 2019-08-29:
Would you be able to provide me with a code snippet (showing how you create an environment and step through it) that I can use to reproduce this issue?

RaveChase, 2019-08-30:
Thx, it turns out the reason that episodes terminate quickly after launching viewer is that the author set the boolean _contact_termination_ equals to _True_ in _dm_control/locomotion/tasks/corridors.py_ and the humanoid will fall in general without a rightful policy. So, this problem should have nothing to do with nconmax and I've solved it.
But the annoying warnings still produce after one statement but I can do nothing.
As for environment, I installed mujoco200 on ubuntu16.04 and dm_control by following your instructions. And created a new py file of which the code is just a paste of **Quickstart** on _dm_control.locomotion_(the link on mainpage).
And the warnings will appear after execute this statement:
`env = basic_cmu_2019.cmu_humanoid_run_walls()`

> WARNING: Logging before flag parsing goes to stderr.
W0829 22:50:16.471426 140720139011840 core.py:299] Pre-allocated contact buffer is full. Increase nconmax above 100. Time = 0.0000.
W0829 22:50:43.107369 140720139011840 core.py:85] Pre-allocated contact buffer is full. Increase nconmax above 100. Time = 0.0000.
W0829 22:50:43.109909 140720139011840 engine.py:323] Physics state is invalid. Warning(s) raised: mjWARN_CONTACTFULL

ChenyangRan, 2020-07-21:
> 事实证明，启动查看器后情节很快终止的原因是作者在_dm_control / locomotion / tasks / corridors.py_中将布尔型_contact_termination_设置为_True_，而类人动物通常会在没有适当政策的情况下_掉队_。因此，这个问题应该与nconmax无关，而我已经解决了。
> 但是在发表声明后仍然会产生恼人的警告，但我无能为力。
> 至于环境，我按照您的说明在ubuntu16.04和dm_control上安装了mujoco200。并在_dm_control.locomotion_（_主页上_的链接）上创建了一个新的py文件，其代码只是**Quickstart**的粘贴。__
> 执行以下语句后将显示警告：
> `env = basic_cmu_2019.cmu_humanoid_run_walls()`
> 
> > 警告：在标志解析之前记录到stderr。
> > W0829 22：50：16.471426 140720139011840 core.py:299]预分配的联系人缓冲区已满。将nconmax增加到100以上。时间= 0.0000。
> > W0829 22：50：43.107369 140720139011840 core.py:85]预分配的联系人缓冲区已满。将nconmax增加到100以上。时间= 0.0000。
> > W0829 22：50：43.109909 140720139011840 engine.py:323]物理状态无效。提出警告：mjWARN_CONTACTFULL

Hi,I meet the same warning as yours when I run the manipulot. Have you solved it? Or it maybe don't matter about the training?