google-deepmind/dm_control #117: If I can't install libglew2.0, can I use dm_control rendering?

Takash7, 2019-10-14:
If I can't install libglew2.0, can I use dm_control rendering? (Ubuntu 16.04, python 3.6.1)

alimuldal, 2019-10-14:
Can you explain why you're unable to install GLEW? Is it because you don't have admin rights on the machine in question? The usual workaround in this situation is to save the shared libraries you need to a directory for which you have write privileges, and then prepend this directory to your `LD_LIBRARY_PATH`.

In fact, MuJoCo comes with its own binaries for GLEW and GLFW3, so it's probably enough to set `LD_LIBRARY_PATH` to `~/.mujoco/mujoco200_linux/bin/`, e.g. `LD_LIBRARY_PATH=~/.mujoco/mujoco200_linux/bin/ python dm_control/suite/explore.py`.

Takash7, 2019-10-15:
I have admin rights and tried the above solution but it didn't work. I think libglew2.0 cannot be used with Ubuntu 16.04.

When I run the following code:
python dm_control/suite/explore.py 
The following error message appears:
glfw.GLFWError: (65543) b'GLX: Failed to create context: BadValue (integer parameter out of range for operation)'

alimuldal, 2019-10-15:
I see, looks as though GLEW 2.0 is not available from the standard repos for Ubuntu <= 16.04. Not sure why the pre-built version is not working for you. Are you able to run the `simulate` binary that's distributed with MuJoCo (`~/.mujoco/mujoco_200/bin/simulate`)?

Takash7, 2019-10-15:
$ ./simulate
MuJoCo Pro version 2.00
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Segmentation fault (core dumped)

Takash7, 2019-10-23:
I solved this problem. An incorrect path was set. Thank you for answering about this question.

closed 2019-10-23 by Takash7