google-deepmind/dm_control #119: WARNING:absl:Cannot set velocity on Entity with no free joint.

Alanliaoyihang, 2019-10-31:
When I try to generate the random motion for the mujoco soccer environment, it appears

Alanliaoyihang, 2019-10-31:
WARNING:absl:Cannot set velocity on Entity with no free joint.

saran-t, 2019-11-01:
Could you please provide a bit more detail on what you're trying to do? This warning is issued when you're trying to set a velocity on static objects.

Alanliaoyihang, 2019-11-01:
what I did is simply run the quick start code for mujoco soccer and try to generate the random motions:
import numpy as np
from dm_control.locomotion import soccer as dm_soccer

# Load the 2-vs-2 soccer environment with episodes of 10 seconds:
env = dm_soccer.load(team_size=2, time_limit=10.)

# Retrieves action_specs for all 4 players.
action_specs = env.action_spec()

# Step through the environment for one episode with random actions.
time_step = env.reset()
while not time_step.last():
  actions = []
  for action_spec in action_specs:
    action = np.random.uniform(
        action_spec.minimum, action_spec.maximum, size=action_spec.shape)
    actions.append(action)
  time_step = env.step(actions)

  for i in range(len(action_specs)):
    print(
        "Player {}: reward = {}, discount = {}, observations = {}.".format(
            i, time_step.reward[i], time_step.discount,
            time_step.observation[i])),
my ubuntu version is 18.04 LT with python3.6

Pavan-Samtani, 2019-11-23:
I too am encountering the same problem. I would like to know how to solve this, or if it can simply be ignored.

closed 2020-01-25 by Alanliaoyihang