google-deepmind/dm_control #122: MuJoCo header and bin path override fails

episodeyang, 2019-12-24:
I tried to install using the following commands but the override does not seem to work. It would be great if there are working examples for these options:

mujoco lib location: `$HOME/.mujoco/mujoco200`

installation script:
```
git clone https://github.com/deepmind/dm_control.git

export MJLIB_PATH=~/.mujoco/mujoco200/bin
cd dm_control
pip install -e . --headers-dir=~/.mujoco/mujoco200/includes
```

I have tried the following and non works:
```
python setup.py --headers-dir=~/.mujoco/mujoco200/includes
```

The only hit of the `headers-dir` in the code base are here: [setup.py#L80](https://github.com/deepmind/dm_control/blob/master/setup.py#L80)

But they don't show up in the commands that are available in the `setup.py` file as if we run `python setup.py -h` shows no such options.

saran-t, 2019-12-24:
I suspect that this is because the `~` is not expanded by Bash in the `--headers-dir=` argument. Try `--headers-dir=${HOME}/.mujoco/mujoco200/includes` instead.

Also, as of yesterday we have uploaded `dm_control` to PyPI so you should be able to `pip install dm_control` as well. If you do this then there's no need to configure MuJoCo's headers path. (You will still need to `export MJLIB_PATH` though.)

episodeyang, 2019-12-24:
thanks Saran, that was not the issue because ~ is interpolated by bash. 

```shell
➜  ~ echo ~
/Users/ge
```

After pip install successfully, running dm_control.suite gives this error
```
OSError: Cannot find MuJoCo library at <my home>/.mujoco/mujoco200/bin.
```
whereas the files are infact there. 

Merry Christmas!

saran-t, 2019-12-24:
The problem with bash expansion is that doesn't work if the argv doesn't begin with `~`.

```echo --headers-dir=~/foo
--headers-dir=~/foo```

This is a bug on our side, we should be applying `os.path.expanduser`to the path. I assume that you've got it installing successfully from PyPI in the end?

The `MJLIB_PATH` environment variable should be the full path of the actual `.so` file itself, not the directory containing the file. (See https://github.com/deepmind/dm_control/blob/master/dm_control/mujoco/wrapper/util.py#L123)

episodeyang, 2019-12-24:
Thanks Saran! just resolved the second problem and indeed it is linked to this issue #6 

The full installation guide that works is here:

1. First set the environments.

```
 31 # <<< conda init <<<
 32 export MUJOCO_GL="glfw"
 33 export MJLIB_PATH=$HOME/.mujoco/mujoco200/bin/libmujoco200.dylib
 34 export MJKEY_PATH=$HOME/.mujoco/mujoco200/mjkey.txt
 35 export LD_LIBRARY_PATH=$HOME/.mujoco/mujoco200/bin:$LD_LIBRARY_PATH
 36 export MUJOCO_PY_MJPRO_PATH=$HOME/.mujoco/mujoco200/
 37 export MUJOCO_PY_MJKEY_PATH=$HOME/.mujoco/mujoco200/mjkey.txt
```

2. install via pip: `pip install dm_control`. Not that installing with `pip install -e .` does not work because if the issue with dynamically generated ctypes #6 

3. the MUJOCO_GL is important for mac os, otherwise you are going to run into a viewer error.
4. the `MUJOCO_PATH` is different for mac os vs linux, because the extension for the library files are difference.
5. In new OSX Catalina, you need to allow the library binary in the securities setting.

saran-t, 2019-12-24:
Thanks for the confirmation: I'll update the README to clarify that the `MJLIB_PATH` should point to the file rather than the directory.

Merry Christmas!

episodeyang, 2019-12-25:
Thanks for your quick response :)

closed 2019-12-25 by episodeyang

kashifzr, 2020-05-26:
I done every but when I am trying to run to code example given with code name with Quick Start with Control suite, following error happen
dm_control.mujoco.wrapper.core.Error:Could not Register licence.

geyang, 2021-06-15:
Now when I install from git installation fails. I think this is related to #6.
```bash
pip install git+git://github.com/deepmind/dm_control.git
```
Gives the following error:
```bash
.mujoco/mujoco200_macos/include/mjdata.h' does not exist.
```

Installation from PyPI works.